FreeCAD/FreeCAD #15760: Stylesheet: Selection color on white has low contrast to the background

maxwxyz, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Related to: https://github.com/FreeCAD/FreeCAD/issues/15746 and https://github.com/FreeCAD/FreeCAD/issues/15749

The selection color and preselection color has no great contrast against the fully white background:
[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36908 (Git)
Build type: Release
Branch: main
Hash: eb842aadffa2615037c45101ac2575fc79438850
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-08-04:
same as https://github.com/FreeCAD/FreeCAD/issues/15746#issuecomment-2267593532

maxwxyz, 2024-08-04:
is it the same color?

luzpaz, 2024-10-26:
is this a dupe of #15746  ?

maxwxyz, 2024-10-26:
Seems similar but I guess both will be fixed by the new core datums anyways.

closed 2024-10-26 by maxwxyz