FreeCAD/FreeCAD #15762: Project: Code of Conduct lacks specific enforcement policies

chennes, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The current https://github.com/FreeCAD/FreeCAD/blob/main/CODE_OF_CONDUCT.md is too broad and too vague about what actions are taken (or should be taken) by community moderators under given circumstances. Current enforcement is somewhat sporadic, and with no clearly-defined "these are the consequences of these actions" moderators are not clearly empowered to take action against individuals who are harming the community through CoC violations.

### Full version info

```shell
N/A
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2024-08-04:
The enforcement section is too broad and too closely related to the forum.  A couple of associated actions could be considered:

1. Review and revise the CoC itself.
   - Articulate our ideals
   - specify what we expect from moderators, contributors, and other participants on our platforms.
   - describe a clear process for escalation and conflict resolution

2. Draft a set of moderation guidelines.  These should be specific enough that moderators know what to do and can point to the guidelines when they take action.  They should be broad enough to let moderators use judgement and assume the best intentions of people.  In other words, the moderation guidelines should be a useful tool and not a binding set of rules.
3. Appoint one or more people to serve as 'ombudsperson' or advocate.  This person would be a resource to help moderators know how to navigate tricky situations.  They would elevate incidents that need more discussion and participation.  They would provide a human name/face when conflict arises.

Some examples of moderation guidelines. (These need to be discussed and carefully worded.  I include them here to give an example of what I think might be helpful)

- "Criticisms of the application or the source code should always be framed in terms of constructive improvements" 
    General criticisms like 'it sucks' are not actionable and only inflame conversation. 

- "Recognize that contributions accepted belong to the greater community." 
    Comments like "XYX feature was written by Bob.  Bob needs to fix it"  create an unfair burden.  They shift blame onto people and discourage shared maintenance and improvement of code.  Any code already accepted into the source tree should be collectively owned and maintained. 
 
- "Criticisms should be accompanied by a willingness to contribute."
- Not everyone is a developer and not everyone should be expected to modify source code. But this doesn't non-developers from contributing.   Someone pointing out problems should be willing to create a GitHub issue and participate in the discussion in a structured and constructive format.  

- "Avoid general criticisms.  Focus on actionable change."
   Bad:  We need a more standard UI
   Good:   XYZ task panel widget alignment isn't consistent with the rest of ABC workbench.
 
- "actionable change should be captured in issues as early as possible."
 The forum, Discord, and other platforms allow for fast and productive discussion but the real work of improving the application is done through GitHub.  
 
- "Don't make great the enemy of good.  Feature and Code improvement is a process."  
FreeCAD improves iteratively.  Criticism of an improvement that falls short of an imagined future state should be discouraged.

yorikvanhavre, 2024-08-05:
I like your proposals @sliptonic. Although these are more guidelines, "things you should care about when writing on the forum", than "rules", I think  they should become enforceable. So basically, nonconstructive criticism would be against the code of conduct and subject to action. Of course this need to be refined because "criticism without a willingness to contribute" could make a first post "illegal" without the person having had the time to demonstrate their willingness. But IMHO what we want here at the end of the day is to define enforceable items.

Since of course these are not huge crimes and should be addressed proportionally (you won ´t get banned because you voiced your criticism a bit too loud), together with these, we should also define the different actions possible. In the "Enforcement" section we could add something like this:

* Forum posts, wiki articles, git comments or any other form of community discussion that infringe this code of conduct will be deleted or the discussion closed. Discussions that derail from a respectful and professional tone and contain personal attacks from an user to another will be immediately closed.

* Authors of personal attacks to other users or other infringements to this code of conduct will be banned temporarily, without warning at first occurrence, and permanently banned on the second.

Also, I would change all "maintainers" texts to "moderators". A moderator should explicitly be a person responsible for applying this code of conduct.

And also there create the ombudsperson role:

* On the different platforms used by the project, abusive behaviors  or any other infringement to this code of conduct should always be reported to the project moderators by using the platform's "report post" or "report content" tools. Moderators will take the action they consider adequate, or discuss the case with the moderation team when unsure.

* Independently of the above, any kind of personal abuse, harassment or any otherwise unacceptable behavior may always be reported to the project ombudsperson. The role of the ombudsperson is to receive, investigate and analyze all complaints received and seek an appropriate response. The ombudsperson will always maintain confidentiality with regard to the reporter of an incident.

* Users that have been attained by an action performed by a moderator, and feel this action was unjust or wrong, may also contact the ombudsperson and have their claim reviewed.

* The ombudsperson will not directly override moderators actions but will always try to solve problems together with the moderation team, and will escalate to project administrators if no solution can be found otherwise.

* The current ombudsperson is indicated on <insert url here> and can be reached via ombudsperson@freecad.org

chennes, 2024-08-05:
> * Forum posts, wiki articles, git comments or any other form of community discussion that infringe this code of conduct will be deleted or the discussion closed. Discussions that derail from a [...]

I'm not sure it''s appropriate to close a discussion because a participant in that discussion couldn't behave: it's a sort of "denial of service" attack, forcing us to close what might otherwise be a productive discussion because of one bad actor. I also think that unprofessional behavior or personal attacks should not be allowed to stand undeleted. IMO, "will be deleted." No "or...".

Roy-043, 2024-08-06:
> A moderator should explicitly be a person responsible for applying this code of conduct.

Depending on interpretation this could be a huge/impossible task, effectively making moderators responsible for the content of all forum posts. We do not have many moderators. Can we ask them to read all forum messages? There are parts of the forum that I currently rarely visit.

> * Someone pointing out problems should be willing to create a GitHub issue and participate in the discussion in a structured and constructive format.

This would ostracize users who do not want to use GitHub for privacy reasons. What are their options?

> unprofessional behavior or personal attacks should not be allowed to stand undeleted.

I agree, but this can be an involved task for moderators. Personal attacks typically are not reported to the moderators, and if they are there has already been a heated exchange on the forum. We do not have 24/7 moderation. The most active moderators are from Europe ATM. So even if a post or topic is reported it may take 8 hours or more before a moderator can take action, which may then involve reading through 10 pages of posts to find out where a discussion derailed, and scrutinizing all subsequent posts.

> * Authors of personal attacks to other users or other infringements to this code of conduct will be banned temporarily, without warning at first occurrence, and permanently banned on the second.

I would prefer to issue a warning first. It should also be pointed out that not every insult/attack is intended as such. There have been users who felt insulted when asked to provide a file for example. And is a post that criticizes some piece of code in an abrasive manner constructive or an attack?

> create the ombudsperson role

What kind of clout ('power') would this person have/need?

sliptonic, 2024-08-06:
Moderator guidelines shouldn't be encoded into the CoC.  They should be documented in a separate place.  They should give the moderators some rationale for effectively taking action.  But that action is always going to be a judgment call.
Without guidelines, moderation is stressful and burns out good people.  

The community wants (and needs) strong moderation.  They want a community where good ideas are discussed freely and trolls are kicked to the curb quickly.

As I said, the examples I gave are just a starting point and need discussion.  It might even help to take a step even farther back.  We can discuss some core values that we want the communities to express and then draft moderation guidelines that we believe will reflect that.

Example:  RESPECT OTHER PEOPLES TIME
Guideline:  "Avoid general criticisms. Focus on actionable change."
Guideline: "Be willing to create a Github issue for a problem you report"
Guideline  "Don't make great the enemy of good. Feature and Code improvement is a process."
Guideline  "Criticisms should be accompanied by a willingness to contribute."
etc.

Example  SHARED OWNERSHIP.  SHARED RESPONSIBILITY
Guideline:  "Don't blame individuals for software limitations and bugs
Guideline: "Don't make great the enemy of good. Feature and Code improvement is a process."

Example:  [DON'T BE A DICK](https://knowyourmeme.com/memes/wheatons-law)
Guideline: Personal attacks are forbidden.
Guideline: Be forgiving of honest mistakes. Learn from them
etc

Does that make sense?  What are the core values that I'm not thinking of?

yorikvanhavre, 2024-08-06:
> I'm not sure it''s appropriate to close a discussion because a participant in that discussion couldn't behave: it's a sort of "denial of service" attack, forcing us to close what might otherwise be [...]

I agree in theory, but the idea was basically to offer moderators an easy action, that simply stops the discussion, without having to "take part" and designate a culprit or do a direct action against a person. I think many moderators feel uneasy with banning or speaking against another community member. This would offer them an additional way to stop the problem from spreading, without taking a personal tone.

Also, in many cases, before you know it, several people begin to argue, and it can be difficult to decide which posts should be deleted, and I think that's another reason why moderation does not work always so well.

But I'm not sold on this, of course.

>  A moderator should explicitly be a person responsible for applying this code of conduct.

> Depending on interpretation this could be a huge/impossible task, effectively making moderators responsible for the content of all forum posts. We do not have many moderators. Can we ask them to read all forum messages? There are parts of the forum that I currently rarely visit.

I agree, this was poorly written and seems to imply moderators have to. I was meaning more that moderators are there basically to follow the code, and cannot be blamed for doing actions, they are not "judges"

Roy-043, 2024-08-12:
I'd like to see some input from other moderators. There is a related discussion item on the 17 August 2024 Developer Meeting Agenda. I'll try to attend that meeting.

This issue was triggered by an item in the 4 August 2024 Developer Meeting. I missed that. It mentions the 'loss of new contributors'. If there are any concrete events I would like to be informed if possible (via forum PM), so that I can learn from them. Thanks in advance.

FEA-eng, 2024-08-17:
Fortunately, most of the incidents we have are not really serious cases of misbehavior as opposed to many other communities. We rarely encounter unacceptable personal attacks. But those would be easy to moderate and we don't even need special rules for them - it's basic netiquette (although it's always good to have those rules stated explicitly in relevant places). The most common issues are related to pointless, non-constructive criticism and those are harder to control. We can't disallow all criticism but we need some rules (as clear as possible even if the line is thin and there will always be questionable cases) at which point to act. For example (similar to what Brad wrote):
1. Criticism is allowed, FreeCAD is not perfect. But make sure to avoid any personal attacks and don't blame a specific dev - it's a community project.
2. To be constructive, always consider how you can contribute. At the very least, you can create a GitHub report - that helps get the issue actually solved. Provide all the relevant information (FreeCAD version, steps to reproduce, screenshots, model files) to make it easier for developers. Of course, if you can fix the issue yourself (submit a PR), it's always more than welcome.
3. Moderators may and will delete posts that present unacceptable behavior (like personal attacks, insulting or swearing, see netiquette) and close the discussions that go too far. This includes violating point 1 but also 2 - if you keep refusing to collaborate in a constructive way or if the discussion evolves into a pointless argument then the moderators will take the necessary actions.

FEA-eng, 2024-08-19:
> I would like to see a rule to clarify that this is not acceptable.

Maybe something like this:
5. FreeCAD, as a community project, is used and developed by people with very different experience, areas of expertise and skills. Everyone is welcome to join the community and we don’t accept criticism based on that. Opinions of less experienced (either with FreeCAD or CAD software in general) users are equally appreciated and respected.

Roy-043, 2024-08-20:
Hi Gene, I hope you are well and that everything is OK in Florida. It still saddens me that you have left the moderation team.

Your version of events however is not correct and I want to clear that up.

The forum topic you refer to discussed moderation standards in general, and, as was made clear in that topic, there was no intention to target you or any other individual moderator.

The criticism was about how we, the forum moderators, including myself, failed to deal with a particularly obnoxious forum member who habitually insulted another contributor (not the developer you mention). IMO this criticism was very much warranted and the moderation team should have taken action much sooner than we did.

Finally, the topic was in a restricted part of the forum only accessible to select group of people, so not public at all.

FEA-eng, 2024-08-21:
Another rule that we could add to cover some more cases:
6. All constructive discussions are welcome but they should happen in the right places. For example, while we do understand that not everyone can have a GitHub account or wants to set it, we would like to keep the coding-related contributions there. So if you think that some FreeCAD code doesn't meet the standards or best practices and should be changed (and you know how so you have a proposed solution), don't just criticize it (see rule 1) and point it out on the forum. Instead, be constructive and create a GitHub issue/PR. Again, moderators will move or close the discussions started in wrong places.

sliptonic, 2024-08-21:
> Another rule that we could add 

This discussion touches on two related but distinct things:  

First the code of conduct.  The CofC should be explicit, simple, and enforced.  It should be (and is) in the root of the source tree.  The CofC is a foundational document.  It's super important.  IMHO, the CofC should articulate what we think is important (civility, cooperation, decency, fairness, etc).  It should state clearly that all perspectives are valued and we won't tolerate shitty behavior.

Second is the moderator _guidelines_   I use that word instead of 'rules'    The guidelines should help the moderators navigate the huge grey areas that will occur in real discussion.  The guidelines should be published and transparent to all but enforcement should be at the discretion of the mods.   They shouldn't be black/white rules used to bash a well-meaning but naive newbie.  They shouldn't be used to extract revenge on a moderator because someone got angry.

Good guidelines help the mod know what to do but don't compel them to any specific course of action.

Does that make sense?

FEA-eng, 2024-08-21:
@sliptonic I see even 3 potential sets of rules/guidelines here:
1) general contributing guidelines (CofC)
2) specific rules for discussions (mostly on the forum) for users
3) guidelines for moderators (but likely also visible to users)

But maybe at least some of them can be combined. However, as opposed to what the original issue itself states, I also agree that CofC within the source code is not necessarily the best place for all of them. So it might be good to keep it short and simple while creating a separate list to be posted in each place where discussions happen. The problem is that we have multiple places like that (including this one) but maybe a pinned post, banner or something similar would work everywhere. A list like mine (of course, it's just a draft and only my suggestion) could be used for that. It covers both 2) and 3) so probably no need to further split it.

yorikvanhavre, 2024-09-02:
> Good guidelines help the mod know what to do but don't compel them to any specific course of action. Does that make sense?

Yes to me. There are two things I wish:

- The mods are better protected, and can more freely apply actions without being blamed
- New users can feel the moderation is in place and happening, and feel safe on the forum

kkremitzki, 2025-07-23:
Revisiting this discussion, I'd say:

1) It's clear the code of conduct needs some revision in general.
2) It also needs specific enforcement measures revision.

These two issues should be considered distinct, although I don't think it makes sense to split them up into two, just yet. For starters, though, I'd say progress on 2) could be most easily spurred on by some research into how other similar projects have handled this, both in terms of practice as well as specific clauses we might want to adopt. If someone would like to contribute to moving this issue forward, I would suggest starting there.

chennes, 2025-07-23:
Blender's Code of Conduct: https://developer.blender.org/docs/handbook/communication/code_of_conduct/
Blender Devtalk forum (development only): https://devtalk.blender.org/
Devtalk rules: https://devtalk.blender.org/t/forum-rules/29170