FreeCAD/FreeCAD #15763: TechDraw: Cannot select anything when views are overlapping

maxwxyz, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I tried to recreate a bug and noticed that I am not able to create dimensions. It happens, when the views overlap, which is the default case when creating a default cube and creating a projection cube with the side views. The view titles in the translated German version are long and this increases the view width so it is entirely on top of the front view where I now cannot select anything. Disabling frames is possible but then I also cannot select geometry.
Would it be possible to click through active views at places where there is no geometry @WandererFan @PaddleStroke ?
I'll submit changes for the translations but this could happen in any language and depending on the part size, scale and distance of the views.

[image]
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-04:
[image]

WandererFan, 2024-08-06:
Use Stack Order to bring the appropriate view to the top.
[image]

benj5378, 2024-09-05:
I tried looking at this. First of all, it seems like the hover event is only forwarded to one QGraphicsItem and it's children. Not other QGraphicsItems directly on the scene. Secondly, it seems like it's only most top child item, that receives the hover event. I'm unsure whether this is built-in Qt behaviour or not.

benj5378, 2024-09-05:
> Secondly, it seems like it's only most top child item, that receives the hover event

Okay, so it looks like the reason for this behaviour is, that the physically hovered item and all its parents receives a hoverevent. This just happens to be the most top item (physically hovered) and all it's parent (the view) at all times.

https://github.com/qt/qtbase/blob/70c47da843a0570784ef1a14d98ccc382425bf1f/src/widgets/graphicsview/qgraphicsscene.cpp#L3804-L3821

Only the top most physically hovered item are getting processed in the `dispatchHoverEvent` (for sending hover event only to it's parents).

https://github.com/qt/qtbase/blob/70c47da843a0570784ef1a14d98ccc382425bf1f/src/widgets/graphicsview/qgraphicsscene.cpp#L3775-L3781

> //
> //  W A R N I N G
> //  -------------
> //
> // This file is not part of the Qt API.  It exists for the convenience
> // of other Qt classes.  This header file may change from version to
> // version without notice, or even be removed.
> //
> // We mean it.
> //

It's actually pretty well coded; how simple the algorithm for a semi complex task (avoiding sending uneeded `hoverEnterEvents`). It's surely beyond my skill level.

Fixing this issue would mean reimplementing this function. There is a pretty big warning with regards to changing their private classes (`QGraphicsScenePrivate`), however, it's at least 13 years ago that `dispatchHoverEvent` was last changed - at least, because git doesn't go further back. So I guess it's safe?

What do you think @WandererFan? Worth to spend time on? Is it safe to reimplement the function?

WandererFan, 2024-10-19:
I have trouble seeing this as a bug.  This seems to be standard behaviour for graphics apps (I used Inkscape to check) - you can't select a item if it is covered by another item.  You have to change the item's z-value (stacking) to get the interesting item to the top.

WandererFan, 2024-10-19:
Does the hover event help us?  We only use that for marking an item as "preselected".  The mouse press event triggers the change to "selected".   Would we select all the items that overlap the mouse press location?  Not sure that is what we want.

We've been asked before for the ability to only select vertexes or edges or...  Maybe this is an opportunity to build a selection processor that would do that and a "superimposed view" mode (?) that would give a list of items at the mouse press location and allow selection from that list.

benj5378, 2024-10-19:
> Does the hover event help us?  We only use that for marking an item as "preselected". 

Most likely the hover event, thay is used internally in Qt to find what to be selected.

WandererFan, 2024-10-19:
Found the request for selection filtering: [Feature Request] TechDraw: Add new option 'Filter of Feature' #6017

WandererFan, 2024-10-19:
"In Inkscape, you actually can."  Didn't know this.  Alt + scroll seems a bit awkward, but it is neat the way it cycles though the overlapping objects.  I don't know off hand if alt + wheel is available.

QGSPage could get a version of: 
QGraphicsScene::wheelEvent([QGraphicsSceneWheelEvent](https://doc.qt.io/qt-6/qgraphicsscenewheelevent.html) *wheelEvent)

Might need to tweak QGVPage::wheelEvent() as well to make sure the event gets to the scene from the widget.

benj5378, 2024-10-19:
> "In Inkscape, you actually can." Didn't know this. Alt + scroll seems a bit awkward, but it is neat the way it cycles though the overlapping objects. I don't know off hand if alt + wheel is available.

@WandererFan alt + click. For each click, you go one layer down.

PaddleStroke, 2025-01-25:
Imo even if it is not technically a bug, it's still a clear usability problem.

>  that the physically hovered item and all its parents receives a hoverevent. This just happens to be the most top item (physically hovered) and all it's parent (the view) at all times.

Maybe I'm missing something as I have not checked the code. But if the view receive the hover, why don't we simple forward the event to the page, then check from the page if anything selectable is actually under the mouse. And if not start the box selection?