FreeCAD/FreeCAD #15764: TechDraw: Projection group dialog saves on abort when dimension tool is open

maxwxyz, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening the projection group and adding views they are saved even when you click cancel, if you invoke the dimension tool with the projection task dialog open.
When clicking cancel, the selected views are saved. If you open the dialog again, you see all checked, removing a checkbox removes the view.
When double clicking again on the projection view, the checkbox is still there, where no view is. You have to uncheck and check again to get the view visible.

Demo:

https://github.com/user-attachments/assets/b68c212f-23b2-42b1-b0f0-fd16254e43a9

@WandererFan @PaddleStroke FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-08-06:
AFAIK, you are only supposed to have one task dialog open at a time, so the "smart" dimension tool should not start if another dialog is open.  All (most?) of the other commands refuse to start if a dialog is active.

PaddleStroke, 2024-08-07:
Correct, the dimension  tool should not be active if another a task is open.