FreeCAD/FreeCAD #15765: TechDraw: Dimension deleted when adding views (also when cancelled)

maxwxyz, 2024-08-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Items under a view (dimensions, balloons,...) get deleted if you open the projection group task dialog.

1. Creating a view from a simple cube, adding a dimension.
2. Double click the view and add another view
3. The dimension gets deleted, regardless if you click CANCEL or OK

IMO it should not get deleted, even when adding a new view and clicking OK.

Demo:

https://github.com/user-attachments/assets/23cd3355-ff98-4138-bcaf-b761a85c7627

@WandererFan @PaddleStroke FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-08-04:
Just to be clear, if a Projection Group is created in the first place and another view is added to it existing dimensions are not affected, only when a single view is 'upgraded' to a Projection Group does it occur.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38394 (Git)
Build type: Release
Branch: main
Hash: 6fda739a14b57da0c7d84230c17933738493d5bf
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * Curves 0.6.42
  * CfdOF 1.27.4
  * fasteners 0.5.25
  * sheetmetal 0.4.23
```

WandererFan, 2024-08-06:
When I do this, the dimension disappears while the new projection group is being built, but it always reappears later on.

PaddleStroke, 2024-08-07:
I have tested and find the same result as WandererFan. The dimension disappear when the view is upgraded to proj group. But as soon as I hit OK it re-appears.

Also after the view is upgraded, you can expand the proj group and view and see that the dimension is still there. So it's just not drawn. We probably need to ask for a redraw after the view is upgraded or something like this.

Syres916, 2024-08-07:
Seems as simple as https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:TD_View_To_ProjGrp_Fix

PaddleStroke, 2024-08-07:
Have you tested it? I think it probably works. Though perhaps if possible we could only repaint the QGraphics to avoid recomputing, which can be long on large models.

Syres916, 2024-08-07:
I dug out a heavy model that didn't have any existing Techdraw elements and created a View then changed to Projection Group and can see no extra performance degradation compared to just creating another view which is the slowest common denominator.

PaddleStroke, 2024-08-07:
I'm currently on a PR for TD so I had a look and found that doing 

    multiView->findParentPage()->requestPaint();
instead of 
    multiView->requestPaint();

Also solve the issue without recomputing. So it's perhaps prefered.

Besides the same needs to be done in the other way around, when proj group is downgraded to view.

PaddleStroke, 2024-08-07:
Ok so the downgrade is much harder to fix apparantly.

https://github.com/user-attachments/assets/3be061ac-b9d3-43b3-b0f6-3d379973c1c7

If you happen to know what can be happening I'm interested! Dunno why the undo is fixing it when recompute is doing nothing...

closed 2024-08-08 by WandererFan