FreeCAD/FreeCAD #15770: Stylesheet: FreeCAD Dark overflow arrow buttons are light

maxwxyz, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The old issue was that the arrows are not legible with the FreeCAD Dark theme, now they are styled light in FreeCAD Dark:
[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-08-05:
These icons cannot be styled atm. Maybe in QT 6.

luzpaz, 2024-08-05:
Does the qss allow the ability to display Qt6 qss formatting as a condition if the user is running Qt6 ?

MisterMakerNL, 2024-08-05:
https://forum.qt.io/topic/119988/qtoolbar-extension-button-stylesheet/2

obelisk79, 2024-08-18:
Those buttons are historically very difficult to style. They don't respond well to stylesheet targeting.

MisterMakerNL, 2024-08-18:
IF we implement what is suggested in the link: https://forum.qt.io/topic/119988/qtoolbar-extension-button-stylesheet/2
Then we can target horizontal and vertical toolbars, which is pretty cool.
Then I can also fix the alignment issue:
[image]

obelisk79, 2024-08-18:
I don't have a problem with those. It's the qToolBar extension symbol that only shows when a toolbar gets truncated.

obelisk79, 2024-08-18:
@MisterMakerNL for the expand buttons you could target them by QToolBar:vertical > QToolButton::menu-arrow {}

MisterMakerNL, 2024-08-18:
I'll try that when I got some time! 
Here is another solution mentioned:
https://stackoverflow.com/questions/30715698/qt-how-to-style-a-qtoolbar-qtoolbutton-overflow-button
Doesn't look to difficult for someone that knows a bit of Freecad coding. @kadet1090 are you done having holiday :P

closed 2025-07-15 by maxwxyz