FreeCAD/FreeCAD #15776: Start page examples are disorganised

qewer33, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are a lot of examples on the start page most of them are outdated and have their own issues. They are also inconsistent with each other. An organization on this front is clearly needed, ideally before 1.0 is shipped.

Some steps that could be taken:
- Same object for the Part Design and Part examples to show the different workflow, also including a TechDraw page
- EngineBlock could be removed in favor of the consistent Part/Part Design examples mentioned above
- FEM examples could be reduced to a single example
- Robot example could be deleted
- Schenkel.stp could be remodeled with the original STEP file also being inside the file to show a "reverse modelling" workflow
- The example names could be made consistent with each other
- Examples could include notes as Draft Annotation objects to inform the user and provide them with the related wiki links

Related PRs, forum posts and links:
https://forum.freecad.org/viewtopic.php?p=535941#p535941
https://forum.freecad.org/viewtopic.php?t=81005
https://github.com/FreeCAD/FreeCAD/pull/15754
https://github.com/FreeCAD/FreeCAD/tree/main/data/examples

### Full version info

```shell
latest main
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qewer33, 2024-08-05:
Mentioning @FEA-eng for the FEM example and @furgo16

FEA-eng, 2024-08-05:
> * FEM examples could be reduced to a single example

Only if absolutely necessary and required by a rule that there should be one example per workbench. We always had a few FEM examples because they show the same case but modeled with very different approaches (3D, 2D and 1D) and yet providing equivalent results. This can be really helpful for users to understand the main workflows. What Bernd mentioned in the referenced forum post has already been done (removal of the "new CalculiX solver" and FEM examples update to take that into account). Initially, there was no 1D example and Bern made and issue to add it so I did it some time ago.

> * Examples could include notes as Draft Annotation objects to inform the user and provide them with the related wiki links

This could be also added as a text document. FEM examples (not the Start page ones but those that are included as scripts) have this:

[image]

FEA-eng, 2024-08-05:
> One of the problems with the FEM examples is that the filenames are too long: it's impossible to see at a glance what the difference between the three very similar-looking examples is supposed to be.

Right, I'll take care of this.

qewer33, 2024-08-05:
> Only if absolutely necessary and required by a rule that there should be one example per workbench. We always had a few FEM examples because they show the same case but modeled with very different approaches (3D, 2D and 1D) and yet providing equivalent results.

But it's all the same simulation (a force acting on a lever) no? So wouldn't it be more organized to have all of them in a single file but grouped via Std::Group folders in the tree view? (a group for 1D, a group for 2D and a group for 3D)

> One of the problems with the FEM examples is that the filenames are too long: it's impossible to see at a glance what the difference between the three very similar-looking examples is supposed to be.

Agreed.

> This could be also added as a text document. FEM examples (not the Start page ones but those that are included as scripts) have this

Yeah this is a better idea.

FEA-eng, 2024-08-05:
@qewer33 I'll see if it works with FEM Analysis container objects. If yes, I'll replace the examples.

MisterMakerNL, 2024-08-07:
Could we use folders?