FreeCAD/FreeCAD #15778: Crash on opening old project

pilhuhn, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was trying to open an old project of mine and FC 0.22.dev immediately crashed on me.

I think this was a project with some of the old assembly workbenches (I guess from the file name), but can't tell. 
I have attached the project file. Even if that WB is invalid right now, it should not crash the binary ...

The mac crash dump for the crashed thread is as follows:

```
System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000030
Exception Codes:       0x0000000000000001, 0x0000000000000030

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [43727]

VM Region Info: 0x30 is not in any region.  Bytes before following region: 4333813712
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10250c000-10252c000    [  128K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libFreeCADApp.dylib           	       0x1056eb6b4 std::__1::__tree_node_base<void*>*& std::__1::__tree<std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::__map_value_compare<Data::MappedName, std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::less<void>, true>, std::__1::allocator<std::__1::__value_type<Data::MappedName, Data::IndexedName>>>::__find_equal<Data::MappedName>(std::__1::__tree_end_node<std::__1::__tree_node_base<void*>*>*&, Data::MappedName const&) + 8
1   libFreeCADApp.dylib           	       0x1056db85c Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) + 832
2   libFreeCADApp.dylib           	       0x1056dceac Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) + 1280
3   libFreeCADApp.dylib           	       0x1056c6820 Data::ComplexGeoData::readElements(Base::XMLReader&, unsigned long) + 1388
4   libFreeCADApp.dylib           	       0x1056c54d0 Data::ComplexGeoData::Restore(Base::XMLReader&) + 864
5   Part.so                       	       0x14fa0b118 Part::PropertyPartShape::Restore(Base::XMLReader&) + 1584
6   libFreeCADApp.dylib           	       0x1055e81a0 App::PropertyContainer::Restore(Base::XMLReader&) + 2512
7   libFreeCADApp.dylib           	       0x1054509d4 App::Document::readObjects(Base::XMLReader&) + 4500
8   libFreeCADApp.dylib           	       0x10544edd4 App::Document::Restore(Base::XMLReader&) + 1400
9   libFreeCADApp.dylib           	       0x105455fe0 App::Document::restore(char const*, bool, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&) + 876
10  libFreeCADApp.dylib           	       0x105681334 App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>&&) + 760
11  libFreeCADApp.dylib           	       0x10567df30 App::Application::openDocuments(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>*, bool) + 2308
12  libFreeCADApp.dylib           	       0x10567d52c App::Application::openDocument(char const*, bool) + 296
13  libFreeCADApp.dylib           	       0x1056b12a4 App::Application::sOpenDocument(_object*, _object*, _object*) + 328
14  libpython3.11.dylib           	       0x102f9fd4c cfunction_call + 124
15  libpython3.11.dylib           	       0x102f48348 _PyObject_MakeTpCall + 332
16  libpython3.11.dylib           	       0x10304a680 _PyEval_EvalFrameDefault + 46152
17  libpython3.11.dylib           	       0x10303e268 PyEval_EvalCode + 220
18  libpython3.11.dylib           	       0x1030a43f4 run_mod + 144
19  libpython3.11.dylib           	       0x1030a7fa0 PyRun_StringFlags + 112
20  libFreeCADBase.dylib          	       0x102c24504 Base::InterpreterSingleton::runString(char const*) + 108
21  libFreeCADApp.dylib           	       0x1056b0ec4 App::Application::sLoadFile(_object*, _object*) + 1256
22  libpython3.11.dylib           	       0x102f9fdec cfunction_call + 284
23  libpython3.11.dylib           	       0x102f48348 _PyObject_MakeTpCall + 332
24  libpython3.11.dylib           	       0x10304a680 _PyEval_EvalFrameDefault + 46152
25  libpython3.11.dylib           	       0x10303e268 PyEval_EvalCode + 220
26  libpython3.11.dylib           	       0x1030a43f4 run_mod + 144
27  libpython3.11.dylib           	       0x1030a7fa0 PyRun_StringFlags + 112
28  libFreeCADBase.dylib          	       0x102c24504 Base::InterpreterSingleton::runString(char const*) + 108
29  StartGui.so                   	       0x143e1b294 StartGui::StartView::fileCardSelected(QModelIndex const&) + 384
30  libQt5Core.5.15.13.dylib      	       0x10637de84 0x106238000 + 1334916
31  libQt5Core.5.15.13.dylib      	       0x106361094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
32  libQt5Core.5.15.13.dylib      	       0x106361094 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
33  CoreFoundation                	       0x1833ce46c __CFRunLoopDoSource0 + 176
34  CoreFoundation                	       0x1833ce1dc __CFRunLoopDoSources0 + 244
35  CoreFoundation                	       0x1833ccdc8 __CFRunLoopRun + 828
36  CoreFoundation                	       0x1833cc434 CFRunLoopRunSpecific + 608
37  HIToolbox                     	       0x18db7019c RunCurrentEventLoopInMode + 292
38  HIToolbox                     	       0x18db6fe2c ReceiveNextEventCommon + 220
39  HIToolbox                     	       0x18db6fd30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
40  AppKit                        	       0x186c2bd68 _DPSNextEvent + 660
41  AppKit                        	       0x187421808 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
42  AppKit                        	       0x186c1f09c -[NSApplication run] + 476
43  libqcocoa.dylib               	       0x1080d44d0 0x1080a8000 + 181456
44  freecad                       	       0x102513204 main + 3356
45  dyld                          	       0x182f660e0 start + 2360

```

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FreeCAD-stickframe
  * ModelRailway
  * LCInterlocking
```

### Subproject(s) affected?

None

### Anything else?

[Brücke-Assembly.FCStd.zip](https://github.com/user-attachments/files/16498846/Brucke-Assembly.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JohnOCFII, 2024-08-05:
Can confirm immediate crash when attempting to open file.

```
OS: macOS 14.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.7.30
```

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [4690]
Path:                  /Applications/Other Apps/FreeCAD-38385.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        Exited process [4678]
User ID:               501

Date/Time:             2024-08-05 09:30:51.1914 -0500
OS Version:            macOS 14.6 (23G80)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Time Awake Since Boot: 1900 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000030
Exception Codes:       0x0000000000000001, 0x0000000000000030

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [4690]

VM Region Info: 0x30 is not in any region.  Bytes before following region: 4298571728
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100370000-100390000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-38385.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libFreeCADApp.dylib           	       0x10354f6b4 std::__1::__tree_node_base<void*>*& std::__1::__tree<std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::__map_value_compare<Data::MappedName, std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::less<void>, true>, std::__1::allocator<std::__1::__value_type<Data::MappedName, Data::IndexedName>>>::__find_equal<Data::MappedName>(std::__1::__tree_end_node<std::__1::__tree_node_base<void*>*>*&, Data::MappedName const&) + 8
1   libFreeCADApp.dylib           	       0x10353f85c Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) + 832
2   libFreeCADApp.dylib           	       0x103540eac Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) + 1280
3   libFreeCADApp.dylib           	       0x10352a820 Data::ComplexGeoData::readElements(Base::XMLReader&, unsigned long) + 1388
4   libFreeCADApp.dylib           	       0x1035294d0 Data::ComplexGeoData::Restore(Base::XMLReader&) + 864
5   Part.so                       	       0x1364e3118 Part::PropertyPartShape::Restore(Base::XMLReader&) + 1584
6   libFreeCADApp.dylib           	       0x10344c1a0 App::PropertyContainer::Restore(Base::XMLReader&) + 2512
7   libFreeCADApp.dylib           	       0x1032b49d4 App::Document::readObjects(Base::XMLReader&) + 4500
8   libFreeCADApp.dylib           	       0x1032b2dd4 App::Document::Restore(Base::XMLReader&) + 1400
9   libFreeCADApp.dylib           	       0x1032b9fe0 App::Document::restore(char const*, bool, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&) + 876
10  libFreeCADApp.dylib           	       0x1034e5334 App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>&&) + 760
11  libFreeCADApp.dylib           	       0x1034e1f30 App::Application::openDocuments(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>> const*, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>>>*, bool) + 2308
12  libFreeCADApp.dylib           	       0x1034e152c App::Application::openDocument(char const*, bool) + 296
13  libFreeCADApp.dylib           	       0x1035152a4 App::Application::sOpenDocument(_object*, _object*, _object*) + 328
14  libpython3.11.dylib           	       0x100e03d4c cfunction_call + 124
15  libpython3.11.dylib           	       0x100dac348 _PyObject_MakeTpCall + 332
[... 2 lines cut ...]
[... cut ...]
36  AppKit                        	       0x193b42cc8 _DPSNextEvent + 660
37  AppKit                        	       0x1943394d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
38  AppKit                        	       0x193b35ffc -[NSApplication run] + 476
39  libqcocoa.dylib               	       0x105f384d0 0x105f0c000 + 181456
40  freecad                       	       0x100377204 main + 3356
41  dyld                          	       0x18fe7b154 start + 2476
```

xtemp09, 2024-08-05:
Looks like #15102 is closely related.

benj5378, 2024-09-01:
This is caused by reading 1771-1784 in Document.xml:
```xml
          <ElementMap count="12">
<Element key="g11;SKT" value="Edge1"/>
<Element key="g11v1;SKT" value="Vertex1"/>
<Element key="g11v2;SKT" value="Vertex2"/>
<Element key="g12;SKT" value="Edge2"/>
<Element key="g12v1;SKT" value="Vertex2"/>
<Element key="g12v2;SKT" value="Vertex3"/>
<Element key="g13;SKT" value="Edge3"/>
<Element key="g13v1;SKT" value="Vertex3"/>
<Element key="g13v2;SKT" value="Vertex4"/>
<Element key="g14;SKT" value="Edge4"/>
<Element key="g14v1;SKT" value="Vertex4"/>
<Element key="g14v2;SKT" value="Vertex1"/>
          </ElementMap>
```

`ComplexGeoData::Restore` calls for `ComplexGeoData::readElements` to restore all these elements, and insert their data into `_elementMap`. However, `_elementMap` has never been initiliazed, being `nullptr`, causing segfault.

4 hours of debugging. You're welcome.

Looks like `_elementMap` is only initialized if the `ElementMap` has attribute `new="1"`, which the `ElementMap` does not. I'm not sure why the code in the first place has logic to call `ComplexGeoData::readElements` without initilization of `_elementMap`, is this a mistake? @chennes 

Does this have potential to affect a lot of users?

bgbsww, 2024-09-01:
Can we get some more details on exactly what freecad version wrote this file?   The element map in the document.xml is weird, and the tag at the start of document.xml is like nothing I've seen:  <Document SchemaVersion="4" ProgramVersion="2021.0205R23662 +3201 (Git)" FileVersion="1" StringHasher="1">

benj5378, 2024-09-01:
I'm not sure if I understand how the ProgramVersion is to be interpreted. But commit number 23662 is from January 2021 and 23662 + 3201 from January 2022. But I have no clue, if I'm right. And back then, there was en ElementMap https://github.com/FreeCAD/FreeCAD/blob/a2f42c65968988f63b3fa21c053b121ff1b7ce7f/src/App/ComplexGeoData.cpp#L446

bgbsww, 2024-09-01:
Uh, follow that link and look at the top of the file:
 *   Copyright (c) 2023 FreeCAD Project Association

That's after 2022.  Blame on line 446 says from #11273 in Nov 2023.

bgbsww, 2024-09-02:
@benj5378 I looked for you on discord and telegram (sorry, I'm one of those non-forum types) and couldn't find a match.  Happy to talk through the evidence in this case, techniques for checking history and files, and generally FC debugging techniques if it is helpful.

JohnOCFII, 2024-09-02:
> @benj5378 I looked for you on discord and telegram (sorry, I'm one of those non-forum types) and couldn't find a match.

Based on the avatar, I think this is @bensay on Discord.

CalligaroV, 2024-09-07:
@pilhuhn I opened a PR with a modification that looks fixing this issue, at least in my tests.
The PR is #16358

Can you too test it and see if this fixes the issue?

This is what I get after opening the file you attached in the description
[image]

Probably it's not 100% what you may expect to see but this is what I get after opening it (most likely because your file has links to other files not attached in the description of the issue)

luzpaz, 2024-09-25:
is this ticket still relevant ?

pilhuhn, 2024-09-25:
Thanks for checking back @luzpaz . FC 1 RC2 does indeed no longer crash and the result after recomputation looks reasonable enough that I can continue working on it (after some fixing of stuff)

luzpaz, 2024-10-15:
Thanks for everyone working on this. Closing

closed 2024-10-15 by luzpaz