FreeCAD/FreeCAD #15780: cMake: No project-level formatting standard

chennes, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Our cMake files use a variety of different formatting standards, and as a project we have not chosen which to standardize on (in general). cMake files should be autoformatted by our pre-commit hook, but to do this we need to agree to the basic standards. Most of this is straightforward, as the creators of cmake have made "official" statements on this. While we aren't *required* to follow those, it seems silly not to. The easiest way to achieve this would be to implement the use of https://github.com/cheshirekow/cmake_format

### Full version info

```shell
N/A
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-10:
Is this still the case ?

chennes, 2025-05-11:
Yes.