FreeCAD/FreeCAD #15781: Mac OS Sonoma: Add on Manager Failed to connect to GitHub

Segran, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a new installation of FreeCAD 21.2 on Mac OS Sonoma 14.5 on a M1 Mac mini where I get this error when trying to run Add on Manager: "Add on Manager Failed to connect to GitHub". The Internet connection is fine when connecting to GitHub using a browser, and I have no proxy.

On my MacBook Pro running Ventura, the Add on Manager works fine.

In addition:
The documentation (seems to be fairly outdated) says that FreeCAD should install a FreeCAD catalog in /users/username/Library/Application Support, but neither in Sonoma or Ventura there is any Library catalog in users - at all. Application Support is located in /Library/Application Support/, but FreeCAD did not install a FreeCD catalog here either.

### Full version info

```shell
ADDON MANAGER DEVELOPER MODE CHECKS
16:04:50  -----------------------------------
16:04:50  -----------------------------------

16:17:17  Addon Manager: Unexpected 0 response from server
16:17:17  Failed to connect to GitHub. Check your connection and proxy settings.
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-05:
Could you please try if you also have an issue with the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

iammattcoleman, 2024-08-05:
@maxwxyz The Addon Manager in 0.21.2 frequently stalls for long periods of times when initially opened, intermittently fails to download metadata for me, and displays the following error when I click on an addon to view its details:
> Could not load README data from URL https://github.co...

The weekly build's Addon Manager works properly.

I'm running the FreeCAD arm64 version on macOS Sonoma 14.6.

maxwxyz, 2024-08-06:
I'm closing this issue if it works on the dev version.

closed 2024-08-06 by maxwxyz

Segran, 2024-08-06:
No. it's exactly the same. As problem persists - please do not close this issue.

maxwxyz, 2024-08-06:
Which version were you using? Please post the full about info.

Segran, 2024-08-06:
FreeCAD_weekly-builds-38419-conda-macOS-arm64-py311.dmg, as off TODAY.

I don’t have more information than that and what I wrote originally, there you fins my OS, hw and the error message. There are no log-files related to FreeCAD.

iammattcoleman, 2024-08-06:
@Segran I apologize for any confusion I might have caused. I think the issues we're seeing with the Addon Manager are similar but not actually the same. Our setup seems practically identical (Sonoma, ARM, recent FreeCAD versions), so we should explore what's different about the environment on our systems.

The "Failed to connect to GitHub." error you're seeing is displayed when a [GET request to the following URL](https://github.com/FreeCAD/FreeCAD/blob/270bb0122ea15bbc562942b9d0756fabb89362c0/src/Mod/AddonManager/addonmanager_workers_startup.py#L95-L97) fails: https://raw.githubusercontent.com/FreeCAD/FreeCAD-addons/master/addonflags.json

Are you able to access that URL?

Also, a bit of a side note: I noticed in this issue's initial description that you said that there's no longer a Library folder within the user home directory on Ventura or Sonoma. That's not the case: the folder is just hidden. You can access it using Finder's "Go to Folder..." feature or via the terminal. Or, if you want, you can make it visible with the following command:
```
chflags nohidden ~/Library
```

Segran, 2024-08-07:
No problem to access the url. 

Found the FreeCAD catalog, thanks for reminding me, I always had hidden files to show on my old Mac, simply forgot to turn it on....