FreeCAD/FreeCAD #15787: Cannot delete External Geometry from the Sketch

leoheck, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

What I am trying to do. I am trying to recover a Sketch, removing everything and adding another geometry. 
But there are some External Geometry that I can never delete. 

Is there any alternative way of doing that?

This circle is External Geometry
[image]

I cannot access the property, it is locked, view only.
[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git) AppImage
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.23
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-05:
Please search for duplicates in the existing issues, this is not the first time you create duplicate issues.
https://github.com/FreeCAD/FreeCAD/issues/15738 and already fixed in tomorrows weekly.

closed 2024-08-05 by maxwxyz

leoheck, 2024-08-05:
@maxwxyz hi, thanks. I did search but I could not find anything.