FreeCAD/FreeCAD #15788: Pressing the delete key with the Export dialog's filename field focused removes the item being exported from the model tree

iammattcoleman, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When exporting an object, if you press the delete key while typing the filename, it deletes the item from the model tree!

The backspace key behaves correctly. For some reason, the model tree still responds to the delete key despite the Export dialog being in the foreground.

This occurs for me with both 0.21.2 and the latest weekly build on macOS Sonoma 14.6 (arm64).

### Full version info

```shell
[code]
OS: macOS 14.6
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Assembly4 0.50.15
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-08-05:
This is odd. On my MacBook Pro, the reverse is true. 

The MBP key marked "delete", which produces a backspace event in FreeCAD and usually doesn't remove things from the tree, works as expected. 

fn-Delete (which produces the actual delete event that will delete things from the tree or sketch etc.) will delete the tree object. 

(It shouldn't, of course; fn-Delete should be doing a forward delete in that field, regardless)

I wonder why the difference. Is this a full Mac keyboard? 

(FWIW I have a US keyboard MBP, though I am in the UK. It produces the US keystrokes as normal though)

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.92.0
  * OpenTheme 2024.7.30
  * fasteners 0.5.25
  * Curves 0.6.42

JohnOCFII, 2024-08-05:
The original issue was against 0.21.  There has been recent work in the area of Macintosh backspace vs. delete, so not surprised behavior is different.  I'm not sure if this is set on a per-workbench or per task item level, so not sure how pervasive the issue really is.

iammattcoleman, 2024-08-05:
@JohnOCFII In the description, I stated that it occurs in both 0.21 and the weekly build. I've updated the version info to match. Sorry for any confusion that might have caused!

When the export (or save) dialog is displayed, it's the only thing that should receive any input. It's quite alarming that you can delete the item you're trying to export because you used the delete key while naming the exported file.

depthoffocus, 2024-08-05:
I don’t have 0.21.x on the Mac but in 0.22-dev it certainly inappropriately
deletes Part workbench parts and Part Design bodies and features.

And — apologies — I amended my comment because I realised Matt was talking
about a physical delete key on a Mac extended keyboard, not the “delete”
key on a MacBook Pro.*

The behaviour I see on my MacBook Pro is consistent with his — the “delete”
key on my keyboard is a backspace key, doesn’t do anything wrong. fn-Delete
is like his physical delete key and that produces the errant behaviour.

** Apple putting “delete” on a backspace key is borderline criminal*

depthoffocus, 2024-08-05:
As an addendum, the problem appears in 0.22-dev regardless of the tree
view type (combined/independent)

iammattcoleman, 2024-08-05:
> I'm not sure if this is set on a per-workbench or per task item level, so not sure how pervasive the issue really is.

It occurs regardless of which workbench is active.

I just verified the behavior with...
* a rectangle (Draft workbench)
* a regular solid (Mesh workbench)
* a sketch
* a spreadsheet

So, it doesn't appear to be tied to the item type.

iammattcoleman, 2024-08-06:
@depthoffocus @JohnOCFII Regarding the key names, I think we're all on the same page but I just want to be 100% clear: I'm referring to the keycode that deletes the character to the right of the cursor in left-to-right languages. PC keyboards label the associated key "delete" or "del". On full-sized Apple keyboards, it's the "forward delete" key above the left arrow key. On Apple laptop keyboards, it's available via fn-"delete".

I am not referring to the backspace key, which Apple confusingly calls "delete" (or "delete (backspace)" in a few headers).

JohnOCFII, 2024-08-06:
> I am not referring to the backspace key, which Apple confusingly calls "delete" (or "delete (backspace)" in a few headers).

Yep -- it's clear.  Hopefully one of the developers that has recently worked on this delete/forward delete/backspace on macOS issue will take a look.  I'm just a Mac user that helps test.

maxwxyz, 2024-08-06:
@wwmayer FYI, you were working on mac Del key iirc.

luzpaz, 2024-08-06:
Related:  
* #9639
* #15057
* #15068 
* #15083
* #15085

luzpaz, 2024-09-30:
What is the status of this ticket ?

depthoffocus, 2024-10-20:
> What is the status of this ticket ?

I have just tested this and I can't get it to happen again. @iammattcoleman can you?

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38991 (Git)
Build type: Release
Branch: main
Hash: ef2156d0f6d116b1642a5531a2d44fc6b7d17042
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.12
  * Curves 0.6.50
```

luzpaz, 2024-10-20:
Thanks. Closing ticket. If reproducible, please notate this ticket and we'll re-open.

closed 2024-10-20 by luzpaz

iammattcoleman, 2024-10-20:
> I have just tested this and I can't get it to happen again. @iammattcoleman can you?
>
> ```
> OS: macOS 14.6.1
> Word size of FreeCAD: 64-bit
> Version: 1.1.0dev.38991 (Git)
> Build type: Release
> Branch: main [...]

I just tested 1.0.0RC2 and was unable to reproduce the issue. (I was unable to test 1.1.0dev.38991 because macOS said the arm64 build was damaged and refused to open it.) Looks like it's been fixed!

luzpaz, 2024-10-20:
Thanks for testing. Check out if the macOS ARM issues is related to https://github.com/FreeCAD/FreeCAD-Bundle/issues/291

iammattcoleman, 2024-10-20:
Yes, that's it. Thanks for the link!