FreeCAD/FreeCAD #15789: Fillet causing a cone to render with a dent

iammattcoleman, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The attached file renders incorrectly with a dent on the side of the upper cone:

[image]

The dent is not present in a 3MF export.

If you set the tip to "exposedCone", the dent goes away. So, it's caused by the fillet.

I tried removing and recreating the fillet and got the same result.

### Full version info

```shell
[code]
OS: macOS 14.6
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Assembly4 0.50.15
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[glass.FCStd.zip](https://github.com/user-attachments/files/16502916/glass.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-08-05:
I've seen things like this before with 0.21, but no dent on:

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.92.0
  * OpenTheme 2024.7.30
  * fasteners 0.5.25
  * Curves 0.6.42

https://github.com/user-attachments/assets/a3c0cd63-c9f6-4fdb-8025-58471228a1ba

iammattcoleman, 2024-08-05:
Ah, you're right. It doesn't happen for me with the weekly build. I should've checked before creating this issue. I'll reopen it if it occurs in the 0.22 release.

closed 2024-08-05 by iammattcoleman