FreeCAD/FreeCAD #15791: Subtractive helix doesn't work if it extends beyond body

laughingrice, 2024-08-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to create a thread:

Create a Tube (or cylinder), then create a additive helix thread to subtract, and subtract thread from tube.
The thread profile can start bellow the tube, but if it goes above, it subtraction fails with error

Cut: Resulting shape is invalid

In the attached file, the cut is broken, if you change the length in AdditiveHelix from 110mm to 100mm, it works fine.
The subtractive helix starts bellow the tube so the effect should be symmetric, but isn't .

[spool_holder_v2.FCStd.zip](https://github.com/user-attachments/files/16503352/spool_holder_v2.FCStd.zip)

### Full version info

```shell
OS: macOS 12.7.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-08-05:
Could this be #14852 ?

maxwxyz, 2024-08-06:
seems to be. @laughingrice can you confirm?

laughingrice, 2024-08-07:
Sorry, missed that one.
Hard to say if this is a duplicate, depends on the internal implementation.
I went down the rabbit hole of references, the first one looks similar. However my attempt at a subtractive helix breaks in a whole different way. It creates a distorted and I believe not-closed part. Creating the helix and then creating a cut breaks completely (empty output), so seems to be going through a different logic 

My solution for now was to create a longer helix and cut the ends off. That did work (a common cut with a box).

maxwxyz, 2024-08-07:
@FlachyJoe can you judge if it is the same problem under the hood as https://github.com/FreeCAD/FreeCAD/issues/14852

chennes, 2024-08-25:
Confirming that this is fixed by #16009

closed 2024-08-25 by chennes