FreeCAD/FreeCAD #15793: Preferences: Headlight direction changes when page is opened, resetting does not reset to default value

maxwxyz, 2024-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening the preference page Display > Light sources and clicking ok without changing anything, changes the headlight direction.

A reset of that page does not reset it to the default (0,0,-1) properly.
On a fresh install it is 0,0,-1. When you just open the preferences and go to the Display>Light source tab the headlight direction changes to something to the top left side. When resetting the preferences it does not change.

It seems like the headlight is controlled by either HeadlightDirection or the vector of HeadlightRotationW, X, Y, Z,... it seems the light adjusts to the parameter which was updated last, either direction or the rotation values. Not sure if this can be fixed for 1.0

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-08-10:
Hello, I've a vision of what the dialog must look like. Here is the video.

https://github.com/user-attachments/assets/1c946393-8999-493d-bece-5c3806ee28fc

What do you think about that?

maxwxyz, 2024-08-11:
@xtemp09 looks good. Does this UI fit a later integration of multiple light sources?
@FreeCAD/design-working-group FYI

xtemp09, 2024-08-11:
> Does this UI fit a later integration of multiple light sources?

I just remade the dialog. If multiple sources are added, it will require another rewriting.

> A reset of that page does not reset it to the default (0,0,-1) properly.

It is because the function of loading direction simply is not called. =)

PR #15877 (2024-08-22): Reimplementation of Light source dialog

@maxwxyz, @chennes, this pull request:

- Loads lightning settings when user opens _Preferences_ (yes, the settings were not loaded before this commit). Closes #15793
- Adds spinboxes to adjust the light direction using XYZ and orientation quaternion notations
- Removes the code that is supposed to adjust settings on-the-fly (it does not work anyway. If I make it work (by moving _View3DSettings_ variable into the class), _NaviCube_ shows up every time user presses the _Apply_ button. I could not find a way to make it go away). The code did not work beca [... cut]
commit ebf02483 (2024-08-22): Reimplementation of Light source dialog (#15877)

* Reimplementation of Light source dialog

Closes #15793.

* Gui: Remove redundant void arguments

---------
closed 2024-08-22 by chennes