FreeCAD/FreeCAD #15794: The "wall length" parameter does not take into account the offset and align

kaiwas, 2024-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Probably something needs to be done with the calculation of the wall length. In the current version, this parameter is calculated from the sketch (base) length. But alignment and offset are not taken into account.

I think the wall calculation should be calculated relative to the middle of the wall. I assume that this is not difficult to do mathematically taking into account alignment and offset.

https://github.com/user-attachments/assets/be5039d7-e20a-4ee2-8e76-496594a74b62

[Wall_lenght.FCStd.del_this_and_zip.zip](https://github.com/user-attachments/files/16506962/Wall_lenght.FCStd.del_this_and_zip.zip)

### Full version info

```shell
any version
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-08-06:
I opened the file again. The wall length is already 4987.19 mm.
And I did not change anything at all. Magic!)

kaiwas, 2024-08-06:
Most likely, the wall length should be calculated based on the middle line of its real geometry.
[image]

kaiwas, 2024-08-06:
In my opinion, it is better to remove this parameter completely if it cannot calculate the length correctly. Because it is better to do it manually than to get a very incorrect result. It was a big surprise for me to learn that the actual length of the wall changes greatly, and the value in this field does not change at all. I am even afraid to imagine what mistakes I could make if I used this parameter when calculating the amount of materials.

paullee0, 2024-08-07:
> Most likely, the wall length should be calculated based on the middle line of its real geometry. (Maybe this is not entirely accurate, but it is definitely more correct than the current approach) ![Scr_24-08-06-11-43-35](https://private-user-images.githubusercontent.com/32273833/355378803-65728d2b-42a4-4bb6-91ca-de1e12d2aae3.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwNDEwMDksIm5iZiI6MTcyMzA0MDcwOSwicGF0aCI6Ii8zMjI3MzgzMy8zNTUzNzg4MDMtNjU3MjhkMmItNDJhNC00YmI2LTkxY2EtZGUxZTEyZDJhYWUzLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDE0MjUwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEyYTk5NzQyZjdhNWZkZDJkMjI0ZWIxM2EyNDNmMGRiMGE1OWNiZjM1MGMzNGMwNTdiZDg5YTdiOThhMzUzYzAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Zb4P49evZBJRYiK_-8HEQ_z_YfhNyc7zHFe8dAqpcSc)

Plus there are individual edge's control of Align, Width, orientation of edges, this would become even more complicated :LOL

kaiwas, 2024-08-07:
> Plus there are individual edge's control of Align, Width, orientation of edges, this would become even more complicated :LOL

Oh, Yes! It's all very interesting )
In general, the "wall length" parameter is useless for walls that consist of different thicknesses. After all, in this case it is more useful to know the lengths of wall fragments that have common parameters.

But nevertheless, if we just need to know the total length of the walls - mathematically it should be calculated along the middle line (as in the picture).
But I can't imagine where this parameter could be used. Perhaps it should simply be removed so as not to mislead the user.
But this parameter could be useful for a wall that consists of only one rectilinear fragment.

paullee0, 2024-08-07:
> Most likely, the wall length should be calculated based on the middle line of its real geometry. (Maybe this is not entirely accurate, but it is definitely more correct than the current approach) [...]
 
I have a look at the code - is the 'length' = (length of wall edges on the 'Right' + on the 'Left') /2  ?

If so, there should be a way that it can be returned (slightly convoluted at the moment though) :D

paullee0, 2024-08-07:
How length of this wall should be counted maybe another problem (https://github.com/FreeCAD/FreeCAD/pull/15350) ?

[wall-alignment-test_Roy_034_a.FCStd.zip](https://github.com/user-attachments/files/16531619/wall-alignment-test_Roy_034_a.FCStd.zip) :D

[image]

paullee0, 2024-08-07:
Another case ... :D

[wall-alignment-test_Roy_034_r5.FCStd.zip](https://github.com/user-attachments/files/16532026/wall-alignment-test_Roy_034_r5.FCStd.zip)

[image]

kaiwas, 2024-08-07:
> I have a look at the code - Is the 'length' = (length of wall edges on the 'Right' + on the 'Left') /2 ?
> 
> If so, there should be a way that it can be returned (slightly convoluted at the moment though) :D

Interesting. How did I get the strange value of 4684.18 mm and then 4987.19 mm?
Something tells me that the current approach is calculating something incorrectly.

kaiwas, 2024-08-07:
It seems that the current code should calculate the "average" line parameter (as in the picture). The logic seems to be like this.
But under certain conditions it does it incorrectly
[image]

PS / I moved one vertex in your sketch a little and everything broke.

kaiwas, 2024-08-07:
But. If the code takes the lengths of the extreme "edges" and divides them in half - then why is the offset not taken into account? Or does it not analyze the real geometry?

paullee0, 2024-08-07:
> > Is the 'length' = (length of wall edges on the 'Right' + on the 'Left') /2 ?
> 
> But. If the code takes the lengths of the extreme "edges" and divides them in half - then why is the offset not taken into account? Or does it not analyze the real geometry?

Currently, the code just take the 'base' Sketch edges, so when there is offset, it is not correct.

I am just thinking whether the 'wall length' = / should be the (length of wall edges on the 'Right' + on the 'Left') /2 mathematically ?    If so, there is something already in code that I can use  (rather than need to 'construct' the center edges  :)

kaiwas, 2024-08-07:
> Currently, the code just take the 'base' Sketch edges, so when there is offset, it is not correct.
>
> I am just thinking whether the 'wall length' = / should be the (length of wall edges on the [...]

The code would work correctly if it considered each segment separately and then summed them up.
but for that there would have to be a way to "identify" a separate segment. Probably it has something to do with the sketch on which the wall is built. (or the line)

yorikvanhavre, 2024-10-11:
> If nobody is working on this, please assign to me and hope to work further on this when this is a gap :)

Done ;)

paullee0, 2024-10-11:
Hmmm, I am not familiar with git / Github, there had been a previous PR on ArchWall pending - https://github.com/FreeCAD/FreeCAD/pull/15195

I should just do below on my pc
```
git fetch upstream
git merge upstream/master
git checkout newBranchName
```
and continue with latest commit for development on this issue ?  or I need to wait for the pending one be merged?

Thanks for direction :)

yorikvanhavre, 2024-10-11:
Since PR #15195 has not been merged yet, I would just add this work to it. I also notice there is a conflict, so you might want to fix the conflict first. Basically, on your machine:

```bash
git pull #  update your main branch
git checkout ArchWall_22_bug-fix_import-ArchSketchObject_add-ArchSketchEdges #  change to your PR branch
git rebase main #  this will rebase your branch onto the most recent code in main. This will fail and produce a conflict
git status  # this will show you which files have conflict
# edit the files with conflicts, look for >>>>> , <<<<< and ====== to find the conflicts, solve the conflicts
git add <the fixed files>
git rebase --continue
git push --force github #  replace 'github' with the name you gave to your own github remote
```

This will give you a clean branch and update your online branch. CI tests will re-run, and you can see if everything is OK after that

After, that, I would just keep adding more commits to the ArchWall_22_bug-fix_import-ArchSketchObject_add-ArchSketchEdges branch, but you can also make another one.

luzpaz, 2024-10-11:
`git merge upstream/master`

@paullee0 JFYI, the branch was renamed to 'main' from 'master'.

paullee0, 2024-10-12:
I do this at the end of the process, hope it is correct :)

```
paullee@fedora:/run/media/paullee/F36_MX500--498G/home00/paullee/Documents/FreeCAD_Macro_Script/FreeCAD$ git push --force --set-upstream origin ArchWall_22_bug-fix_import-ArchSketchObject_add-ArchSketchEdges
```

```
Username for 'https://github.com': paullee0 xxx
Password for 'https://paullee0 xxx ': 
Enumerating objects: 11, done.
Counting objects: 100% (11/11), done.
Delta compression using up to 4 threads
Compressing objects: 100% (6/6), done.
Writing objects: 100% (6/6), 1.11 KiB | 37.00 KiB/s, done.
Total 6 (delta 5), reused 0 (delta 0), pack-reused 0 (from 0)
remote: Resolving deltas: 100% (5/5), completed with 5 local objects.
To https://github.com/paullee0/FreeCAD.git
 + 03b3db644a...e1f6ff65a5 ArchWall_22_bug-fix_import-ArchSketchObject_add-ArchSketchEdges -> ArchWall_22_bug-fix_import-ArchSketchObject_add-ArchSketchEdges (forced update)
branch 'ArchWall_22_bug-fix_import-ArchSketchObject_add-ArchSketchEdges' set up to track 'origin/ArchWall_22_bug-fix_import-ArchSketchObject_add-ArchSketchEdges'.

```

Just keep it this way, would separate branch for further commit or I have the feeling I'll mess up git again, again and again :D

paullee0, 2024-10-13:
OK, first draft fix below :) Special Remarks to this Fix :
1.  This _excludes_ **PropertySet** support which is developed in separate branch
2.  This includes / is based on  PR #15195 for convenience
3.  This includes some groundwork for some other features, e.g.
SketchArch add-on
- Different Wall joint types
- Arch Objects setting-out from corner of room (in addition to wall segment Axis supported currently)
- etc.

[Test_ ArchWall_ 42_ BugFix_Length.FCStd.zip](https://github.com/user-attachments/files/17352823/Test_.ArchWall_.42_.BugFix_Length.FCStd.zip)

**Before Fix / Recompute**
[image]

**After Fix / Recompute - Align Left/ Right/ Center**

[image]

[image]

[image]

paullee0, 2024-10-13:
Can  you help testing and put these diagrams in the wiki for ease of understanding please?  :) 

> Most likely, the wall length should be calculated based on the middle line of its real geometry. (Maybe this is not entirely accurate, but it is definitely more correct than the current approach) [...]

> It seems that the current code should calculate the "average" line parameter (as in the picture). The logic seems to be like this. But under certain conditions it does it incorrectly ![2024-08-07 20_35_2125](https://private-user-images.githubusercontent.com/32273833/355923938-7bf0ae72-f0ad-4859-913b-b8163413f1c7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3ODcwODEsIm5iZiI6MTcyODc4Njc4MSwicGF0aCI6Ii8zMjI3MzgzMy8zNTU5MjM5MzgtN2JmMGFlNzItZjBhZC00ODU5LTkxM2ItYjgxNjM0MTNmMWM3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEzVDAyMzMwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNjY2YzYjk4ODMwZDQzMTkyZmQzZDE5NDk3MWFjZjFmMjFiYzY1YWM5NWM0ZDY3Yzc0NzI3NjhkYTM3MWVjZDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.RUQFaXv43nB4FcXc58dp5QYi7KaIhmqIvryf8mQFcWM)
> 
> PS / I moved one vertex in your sketch a little and everything broke.

paullee0, 2024-10-13:
@kaiwas  Can  you help testing and put these diagrams in the wiki for ease of understanding please?  :) 

> Most likely, the wall length should be calculated based on the middle line of its real geometry. (Maybe this is not entirely accurate, but it is definitely more correct than the current approach) [...]

> It seems that the current code should calculate the "average" line parameter (as in the picture). The logic seems to be like this. But under certain conditions it does it incorrectly ![2024-08-07 [...]

paullee0, 2024-10-13:
I have closer look and realise corner case here -  The length of those internal wall or those has point on line coincident on other walls, if the latter are Left or Right Align, would be 1/2 of the latter's Wall Width shorter or longer :(  Any suggestion?

[image]
[image]

kaiwas, 2024-10-13:
> Can you help testing and put these diagrams in the wiki for ease of understanding please? :)

I don't quite understand how I can test it?
Have you implemented a way to calculate the length of a wall in SketchArch?

As for pictures and posting them to the wiki. I've never done this and don't know how to do it. Besides, the wiki is too serious and I think the pictures need to be improved before posting them to the wiki.

paullee0, 2024-10-13:
> I don't quite understand how I can test it? Have you implemented a way to calculate the length of a wall in SketchArch?
>
> As for pictures and posting them to the wiki. I've never done this and don't [...]

Oops, I forgot to upload the revised files in the Mod (BIM, Draft), uploaded now above.  See if you can test if it break any of your existing model or other other of regression  (regardless the difference found above).  No worry about the wiki, feel free if  you are interested and would like to involve :)  Thanks.

kaiwas, 2024-10-13:
> No worry about the wiki, feel free to help if you are interested and would like to involve

It's not that I'm embarrassed. It's just that my English is lame on both legs. I might write something bad.

I'll check the files.

kaiwas, 2024-10-13:
Did I understand the task correctly? (I unpacked appimage).

Replace this file
> * BIM: ArchWall.py

I add these files to the /mod/DRAFT folder (they are not there by default)
> *Draft : offsets.py, intersections.py

In this case, my walls are not built (some kind of error)

https://github.com/user-attachments/assets/c45cb84d-7f7d-4cec-b7b4-209c37a847d1

```

10:58:48  pyException: Traceback (most recent call last):
  File "/home/kaiwas/PROGI/squashfs-root/usr/Mod/BIM/ArchWall.py", line 299, in execute
    extdata = self.getExtrusionData(obj)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kaiwas/PROGI/squashfs-root/usr/Mod/BIM/ArchWall.py", line 1110, in getExtrusionData
    wNe2 = DraftGeomUtils.offsetWire(wire, dvec,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'TypeError'>: offsetWire() got an unexpected keyword argument 'wireNedge'
10:58:48  Wall: offsetWire() got an unexpected keyword argument 'wireNedge'
```

kaiwas, 2024-10-13:
This is not related to the topic under discussion. It is more about sketchchart.
If we use some large wall offset, then the windows and doors we place are torn off from the wall, because they are tied to the sketch, not to the wall (this is a disadvantage of this approach).
[image]
[image]

Now on to the topic.
Do we have the ability to get the areas of the designated figures at any stage (regardless of the openings)?
[image]
Then we could get the length of each segment from the area and width values ​​(area divided by width equals length)

paullee0, 2024-10-13:
Thanks for testing:)

> Replace this file
> 
> > * BIM: ArchWall.py

This is correct 

> I add these files to the /mod/DRAFT folder (they are not there by default)
> 
> > *Draft : offsets.py, intersections.py

For these 2 files, find below folder, backup originals, and copy these 2 new files in.
- FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311/squashfs-root/usr/Mod/Draft/draftgeoutils

> UPD/ I got it) I didn't turn on sketcharch))) sorry
 
For testing the the current fix, I think the SketchArch add-on needs to be disabled ... I haven't incorporated the latest PropertySet support and this fix altogether, hopefully after this one is done.  Thanks.

paullee0, 2024-10-13:
> This is not related to the topic under discussion. It is more about sketcharch. If we use some large wall offset, then the windows and doors we place are torn off from the wall, because they are tied to the sketch, not to the wall (this is a disadvantage of this approach). ![Снимок экрана_20241013_112247](https://private-user-images.githubusercontent.com/32273833/376019354-97828acb-4495-4ab0-a258-c80930d6e3a4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg4MDk3MDcsIm5iZiI6MTcyODgwOTQwNywicGF0aCI6Ii8zMjI3MzgzMy8zNzYwMTkzNTQtOTc4MjhhY2ItNDQ5NS00YWIwLWEyNTgtYzgwOTMwZDZlM2E0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEzVDA4NTAwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThlNzA4YjcxZTgwMjgwOGZlOTIxNTM2MGY5MmVhOTIwMDRlYjM2MDdjNTFjMTBmMzY2OWE3ODliOTZhYjYwZjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.DUneBvElCJLuDllhsVAIu1evMbcI1YY1w2_eMJ0whrs) ![Снимок экрана_20241013_112303](https://private-user-images.githubusercontent.com/32273833/376019356-d091bb67-66e7-4c09-b4f7-a8b9341d254e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg4MDk3MDcsIm5iZiI6MTcyODgwOTQwNywicGF0aCI6Ii8zMjI3MzgzMy8zNzYwMTkzNTYtZDA5MWJiNjctNjZlNy00YzA5LWI0ZjctYThiOTM0MWQyNTRlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEzVDA4NTAwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBiMjU5ODhjZWY1Njg4MTk1ZmFiNDBmMzQzYmUxMDM4ZjU3M2Y2YzkxNzMxNWJhMTdmZjAyMzU0NGIxZmQ1NjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.UaX6UT9PAgAYKlNhkxxqmC__nABDs2KNrK2AjTNcRC4)
> 

There is an 'Attachment Alignment' property added in Window, if it is Wall Left or Wall Right, it should stick somewhere to the Wall surface.  If not there is some kind of regression or bug.  Please report in the Forum or the [SkechArch github](https://github.com/paullee0/FreeCAD_SketchArch).  Thanks :)

paullee0, 2024-10-13:
> Now on to the topic. Do we have the ability to get the areas of the designated figures at any stage (regardless of the openings)? ![Снимок экрана_20241013_112537](https://private-user-images.githubusercontent.com/32273833/376019430-c2dfbef2-7287-425a-bc82-12595c462494.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg4MDk3MDcsIm5iZiI6MTcyODgwOTQwNywicGF0aCI6Ii8zMjI3MzgzMy8zNzYwMTk0MzAtYzJkZmJlZjItNzI4Ny00MjVhLWJjODItMTI1OTVjNDYyNDk0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEzVDA4NTAwN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNlOTkxN2YxNmEyNmMyNTI2MzNkMmY4MThmYWFmNjk4MTA0ZTI0N2U3Y2Y1OWIxYzhjOGIwNjI2MzdlOWJlODgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.NngxpRc_U8qojiOlk5WMWoQynr07ds02pNEKnMiVtSo) Then we could get the length of each segment from the area and width values ​​(area segment divided by width equals length segment)

Yes, the ground work I mentioned return every connected edges of the ArchWall, so that's why I can calculate the length of them, add them together and do an average.  Just priority, time, available are consideration :D

kaiwas, 2024-10-13:
> For testing the the current fix, I think the SketchArch add-on needs to be disabled ... I haven't incorporated the latest PropertySet support and this fix altogether, hopefully after this one is done. Thanks.

Thank you! I replaced the files correctly.
I have not checked the correctness of the calculations, but I see that the length changes depending on the offset, wall width and alignment. This is just great!

I will be able to check the correctness of the calculations another time (now my brain refuses to make calculations)

[image]
[image]
[image]

paullee0, 2024-10-13:
> Yes, the ground work I mentioned return every connected edges of the ArchWall, so that's why I can calculate the length of them, add them together and do an average. Just priority, time, available [...]

BTW, there is a Horizontal Area in the Property, I think that gives you the total ArchWall footprint area, see if that's already good for you :)

kaiwas, 2024-10-13:
> BTW, there is a Horizontal Area in the Property, I think that gives you the total ArchWall footprint area, see if that's already good for you :)

Yes. This data also changes (and does not depend on the presence of doors). I do not need this parameter, because the volume is usually more important to me. But it is still useful for someone.

Now I have to calculate the volume through the expressions wall.shape.volume If such a property appeared in the wall itself, it would be great. But this is a topic for another ticket)

paullee0, 2024-10-14:
Suggest to provide a more detailed property tip, drafted below for reference and comment -

" The length of the wall input by user, if the wall is based on an unconstrained sketch with a single edge, or on a Draft Wire with a single edge.  It becomes output and read only, if the wall is based on constrained sketch or DWire with multiple edges.  The length returned is the medium of the ArchWall if each segment of wall is of different Width, Align and/or Offset; would have minor deviation or complication in measurement if the sketch is complex e.g. internal wall having a T-junction or self-intersecting wall segment.  Use Horizontal Area property which return the footprint area and thus better representation of the wall without complication in calculation of 'length' in above corner-cases."

ArchWall.py as revised (and with other minor code refinement)

[ArchWall.py.txt](https://github.com/user-attachments/files/17367734/ArchWall.py.txt) 
(remove .txt)

paullee0, 2024-10-17:
Before I submit a PR :

Comment on the above property tip from native English speaker?  

And confirmation of the calculation is more accurate now?

Thanks.

kaiwas, 2024-10-24:
> And confirmation of the calculation is accurate enough now?
> 
> Thanks.

I can't help with the language issue.
But the correct calculations at first glance are much better than the original version. I think in any case it will be better than now.
The request to merge with the main branch - later we will test it with many users. The current approach is still useless because it is categorically wrong.

yorikvanhavre, 2024-10-24:
It's a long text for a tooltip, no? Maybe better to redirect the user to the documentation and explain better there?

closed 2024-11-11 by yorikvanhavre