FreeCAD/FreeCAD #15797: Sketcher doesn't handle autoconstraints for slot tool. Also, a crash.

olvll, 2024-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As previously discussed on https://forum.freecad.org/viewtopic.php?p=774199 (I've also posted video there to illustrate the problem).

When trying to draw a slot between two points that are already constrained horizontally or vertically, the Sketcher seemingly fails to create the slot and displays two messages:

`Autoconstraints cause redundancy. Removing them
Redundant constraint is not an autoconstraint. No autoconstraints or additional constraints were added. Please report!`

At this point the sketch is marked as invalid (orange). Two ways out of this state:

1. Draw a random line somewhere. Now the previously invisible slot appears, and I can add missing constraints manually.
2. Close the sketch and open it again, and proceed like in 1.

On the other hand, if I try to select elements of the sketch while they are still marked in orange, FreeCAD crashes with a segmentation fault error:

`Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7fe6d1242520]
#1  0x7fe6984cbd97 in SketcherGui::EditModeInformationOverlayCoinConverter::convert(Part::Geometry const*, int) from /usr/lib/freecad-daily-python3/lib/SketcherGui.so+0x18d7
#2  0x7fe6984d7c7d in SketcherGui::EditModeCoinManager::processGeometryInformationOverlay(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const> > > const&) from /usr/lib/freecad-daily-python3/lib/SketcherGui.so+0xcd
#3  0x7fe6984d7f15 in SketcherGui::EditModeCoinManager::processGeometryConstraintsInformationOverlay(Sketcher::GeoListModel<std::unique_ptr<Sketcher::GeometryFacade const, std::default_delete<Sketcher::GeometryFacade const> > > const&, bool) from /usr/lib/freecad-daily-python3/lib/SketcherGui.so+0x35
#4  0x7fe6984fa888 in SketcherGui::ViewProviderSketch::draw(bool, bool) from /usr/lib/freecad-daily-python3/lib/SketcherGui.so+0x68
#5  0x7fe6984f4399 in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /usr/lib/freecad-daily-python3/lib/SketcherGui.so+0x529
#6  0x7fe6d47e62be in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x15e
#7  0x7fe6d0c4db47 in SoEventCallback::handleEvent(SoHandleEventAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xb7
#8  0x7fe6d0a76095 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#9  0x7fe6d0c10908 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#10  0x7fe6d0c57660 in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x50
#11  0x7fe6d0c6c66c in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#12  0x7fe6d0a76095 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#13  0x7fe6d0c10908 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#14  0x7fe6d0c57660 in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x50
#15  0x7fe6d0c6c66c in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#16  0x7fe6d0a76095 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#17  0x7fe6d0a7fa9b in SoHandleEventAction::beginTraversal(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xab
#18  0x7fe6d0a766e8 in SoAction::apply(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xc8
#19  0x7fe6d0c25559 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x79
#20  0x7fe6d477d0e2 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x22
#21  0x7fe6d478e649 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x59
#22  0x7fe6d479a57d in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x2d
#23  0x7fe6d479c201 in Gui::CADNavigationStyle::processSoEvent(SoEvent const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x211
#24  0x7fe6d479aa46 in Gui::NavigationStyle::processEvent(SoEvent const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x66
#25  0x7fe6d47ba51e in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x6e
#26  0x7fe6d4780b9d in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x1cd
#27  0x7fe6d1cb9b9a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0xaa
#28  0x7fe6d296c702 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x72
#29  0x7fe6d2974364 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#30  0x7fe6d449c868 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x98
#31  0x7fe6d1cb9e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#32  0x7fe6d2972e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#33  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c8d40) [0x7fe6d29c8d40]
#34  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cbfd5) [0x7fe6d29cbfd5]
#35  0x7fe6d296c713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#36  0x7fe6d449c868 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x98
#37  0x7fe6d1cb9e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#38  0x7fe6d2141307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x697
#39  0x7fe6d2116a2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#40  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7fe6cbd57d6e]
#41  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7fe6d0719d3b]
#42  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab2b8) [0x7fe6d076f2b8]
#43  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fe6d07173e3]
#44  0x7fe6d1d130b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#45  0x7fe6d1cb875b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#46  0x7fe6d1cc0cf4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#47  0x7fe6d442209b in Gui::Application::runApplication() from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x8db
#48  freecad-daily(+0x6d54) [0x559654b47d54]
#49  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fe6d1229d90]
#50  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fe6d1229e40]
#51  freecad-daily(+0x7ea5) [0x559654b48ea5]
`

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38392 (Git)
Build type: Release
Branch: main
Hash: c87248e373c373174e3e12f0ae245b6c08a7e560
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Curves 0.6.42
  * A2plus 0.4.65
  * sheetmetal 0.4.23
  * fasteners 0.5.25
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-08-07:
I have seen the autoconstraint issue with 38385 -- the odd thing is in some circumstances it *does* manage to remove the redundant constraint, and tells you.

It typically will occur if you try to draw a slot where both of the endpoints are on an axis. You can resolve the problem by e.g. removiong 

I haven't seen the crash in 38385, but I will test in 38419.

depthoffocus, 2024-08-07:
Ahh, I see the specific nuance here.

OK so if I do the minimal possible situation to trigger this and just put two points in, constrain them to be vertically aligned, and draw a slot, I get the error and the slot doesn't appear until that vertical constraint is removed. If I remove the vertical constraint, the slot appears!

A fuller example with some more geometry is like you say:

https://github.com/user-attachments/assets/14605d77-fdf8-416b-9761-f98eb0a55c7d

 Though I've not seen a crash.

olvll, 2024-08-07:
The crash can be seen in my video that I originally included in the forum post. Here it is again:

https://www.youtube.com/watch?v=fLPdb_V8NFI

It happens when the sketch is in the "invalid" state and I select some elements by mouse dragging. Clicking to select them individually does not make FreeCAD crash.

depthoffocus, 2024-08-07:
> select some elements by mouse dragging

OK -- yes, I can replicate that as well!

Relevant thread crash on Mac 38385, details as above:

```
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [21314]

VM Region Info: 0 is not in any region.  Bytes before following region: 4313726976
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1011e4000-101203000    [  124K] r-x/r-x SM=COW  /Applications/FreeCAD (38385).app/Contents/Resources/bin/freecad

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   SketcherGui.so                	       0x11eaf0d07 void SketcherGui::EditModeInformationOverlayCoinConverter::addUpdateNode<SketcherGui::EditModeInformationOverlayCoinConverter::NodePolygon<(SketcherGui::EditModeInformationOverlayCoinConverter::CalculationType)5>>(SketcherGui::EditModeInformationOverlayCoinConverter::NodePolygon<(SketcherGui::EditModeInformationOverlayCoinConverter::CalculationType)5> const&) + 103
1   SketcherGui.so                	       0x11eaffcfd SketcherGui::EditModeCoinManager::processGeometryInformationOverlay(Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>> const&) + 253
2   SketcherGui.so                	       0x11eaffa83 SketcherGui::EditModeCoinManager::processGeometryConstraintsInformationOverlay(Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>> const&, bool) + 115
3   SketcherGui.so                	       0x11eb26b47 SketcherGui::ViewProviderSketch::draw(bool, bool) + 167
4   SketcherGui.so                	       0x11eb21f41 SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) + 3921
5   libFreeCADGui.dylib           	       0x1032853b7 Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 503
6   libCoin.80.0.2.dylib          	       0x105bd13b5 SoEventCallback::handleEvent(SoHandleEventAction*) + 197
7   libCoin.80.0.2.dylib          	       0x1059bed89 SoAction::traverse(SoNode*) + 105
8   libCoin.80.0.2.dylib          	       0x105b87ef3 SoChildList::traverse(SoAction*, int, int) + 131
9   libCoin.80.0.2.dylib          	       0x105bdb1c5 SoGroup::doAction(SoAction*) + 101
10  libCoin.80.0.2.dylib          	       0x105bf6858 SoSeparator::handleEvent(SoHandleEventAction*) + 40
11  libCoin.80.0.2.dylib          	       0x1059bed89 SoAction::traverse(SoNode*) + 105
12  libCoin.80.0.2.dylib          	       0x105b87ef3 SoChildList::traverse(SoAction*, int, int) + 131
13  libCoin.80.0.2.dylib          	       0x105bdb1c5 SoGroup::doAction(SoAction*) + 101
14  libCoin.80.0.2.dylib          	       0x105bf6858 SoSeparator::handleEvent(SoHandleEventAction*) + 40
15  libCoin.80.0.2.dylib          	       0x1059bed89 SoAction::traverse(SoNode*) + 105
16  libCoin.80.0.2.dylib          	       0x1059c9a0d SoHandleEventAction::beginTraversal(SoNode*) + 157
17  libCoin.80.0.2.dylib          	       0x1059beada SoAction::apply(SoNode*) + 362
18  libCoin.80.0.2.dylib          	       0x105ba2db1 SoEventManager::actuallyProcessEvent(SoEvent const*) + 97
19  libFreeCADGui.dylib           	       0x10321a94c SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) + 28
20  libFreeCADGui.dylib           	       0x103221693 SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) + 131
21  libFreeCADGui.dylib           	       0x10322f0ef Gui::NavigationStyle::processSoEvent(SoEvent const*) + 79
22  libFreeCADGui.dylib           	       0x103235f6d Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) + 1389
23  libFreeCADGui.dylib           	       0x10322ee14 Gui::NavigationStyle::processEvent(SoEvent const*) + 180
24  libFreeCADGui.dylib           	       0x10325279f Gui::View3DInventorViewer::processSoEvent(SoEvent const*) + 175
25  libFreeCADGui.dylib           	       0x103212c27 SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) + 247
26  libQt5Core.5.15.13.dylib      	       0x104ef8273 QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) + 157

```

JohnOCFII, 2024-08-07:
Could also be related to: https://github.com/FreeCAD/FreeCAD/issues/15494

maxwxyz, 2024-08-13:
Does this work with the stable 0.21.2 version?

depthoffocus, 2024-08-13:
> Does this work with the stable 0.21.2 version?

Due to having an unintentionally airgapped machine I can't get a screenshot uploaded (or get my About info) but with 0.21.2 on Windows (x86-64), the slot does correctly appear when drawn as above -- you get an overconstrained sketch as you should.

FEA-eng, 2025-09-20:
@maxwxyz This is pretty bad, perhaps it could be a blocker for 1.1 (unless it can't be fixed in time anyway). Related issue: #17074

Basically, just drawing a slot with both points on the X axis shows it as underconstrained but then leaving the sketch turns it into redundant constraints error.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

AjinkyaDahale, 2025-12-15:
Cannot reproduce with
```
OS: Ubuntu 25.10 (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.2.0dev.44672 (Git)
Build date: 2025/12/13 04:03:31
Build type: Unknown
Branch: part-sk-restricted-curves
Hash: c24c4680d826cbc9b8fd09f0ecfdb507639eade0
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

For some reason I'm not able to run the snap at this moment, but in the above branch I don't believe there's anything affecting the current issue.

PaddleStroke, 2025-12-31:
So I have made progress on this bug.

- First I noticed that 'Snap to object' is NOT working on the axis. So when you draw a slot on one axis, the geometries are initially NOT perfectly horizontal.

Now about the solver redundancy bug, it actually depends on snapToObject working or not.
1 - Without SnapToObject (ie current broken situation):
- With EigenSparseQR : Failing to detect
- With EigenDenseQR : Failing to detect

2 - With SnapToObject FIXED and enabled:
- With EigenSparseQR : Failing to detect
- With EigenDenseQR : Works!!

Continuing the investigation

PaddleStroke, 2025-12-31:
Alternative / complement idea, is to enforce DenseQR when there are less than an arbitrary number of parameters, say 200 for example. This way for small sketches where the compute difference is not great we use the more robust algorithm.

```
int System::diagnose(Algorithm alg)
{
    ....
    hasDiagnosis = true;
    dofs = pdiagnoselist.size();

    // Heuristic: Use DenseQR for small to medium systems to avoid SparseQR rank issues.
    // SparseQR is known to fail rank detection on specific geometric structures (e.g. aligned slots).
    // 200 parameters roughly corresponds to ~100 points/curves, covering most complex sketches
    // where stability is preferred over pure O(N) performance.
    // See: https://github.com/FreeCAD/FreeCAD/issues/15797
    if (qrAlgorithm == EigenSparseQR && dofs ) < 200) {
        qrAlgorithm = EigenDenseQR;
    }
```

AjinkyaDahale, 2025-12-31:
I did involve myself with some Eigen a few years ago, but now with much benefit.

For 200 points, we are looking at 40,000 element matrix with denseQR as opposed to, maybe, 2,000 elements in sparse notation (though for the QR factorization the density anyway increases). I would also not be surprised if denseQR also has more computations and hence a harder time computing rank.

Are we getting a higher or lower rank in this case? Also, is it possible to test for rank estimation issues directly without involving freecad?

PaddleStroke, 2025-12-31:
@AjinkyaDahale if you want to test I made a PR : https://github.com/FreeCAD/FreeCAD/pull/26559

I do not notice any slow down in a sketch close to the limit. So I think it becomes too slow when dof is greater.
Let's discuss on that PR btw

AjinkyaDahale, 2026-01-07:
Could this be caused by an error in the implementation of eigen in Windows?