FreeCAD/FreeCAD #15799: Stylesheet: Property view rows are always visible on macOS

maxwxyz, 2024-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On macOS the different colored rows in the property view are always visible, even when nothing is selected or no object is in the document. This does not happen on Windows.

[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-06:
When selecting an empty body, the rows are rendered differently where there is actually a property:
[image]

maxwxyz, 2024-08-20:
@MisterMakerNL any news on this?

MisterMakerNL, 2024-08-20:
I cannot test atm.
Coul;d you check opentheme if it is also there?

luzpaz, 2024-08-20:
Hey @johnocfii can you test on macos ?

maxwxyz, 2024-08-24:
@MisterMakerNL same issue with OpenTheme

@kadet1090 @obelisk79 FYI

obelisk79, 2024-08-24:
> @MisterMakerNL same issue with OpenTheme
> 
> @obelisk79 FYI

I am unable to recreate this on OpenTheme

MisterMakerNL, 2024-08-24:
@JohnOCFII Do you have most current opentheme, since these rows are recently added by @kadet1090 as previous they couldn't be style which resulted in inverse problem of never having a background.

JohnOCFII, 2024-08-24:
> @JohnOCFII Do you have most current opentheme, since these rows are recently added by @kadet1090 as previous they couldn't be styled which resulted in inverse problem of never having a background.

Yes, I've got the most recent version of OpenTheme.  What I don't know is if some combination of overlay or transparent window control could influence this behavior.

Odd that this Add-on description window doesn't show the version number, but the main Add-on list of items does.
[image]

kadet1090, 2024-08-24:
This is strictly related to the MacOS QtStyle probably, no way to do anything about it without changing the style.

maxwxyz, 2024-08-25:
@kadet1090 @MisterMakerNL would https://github.com/FreeCAD/FreeCAD/pull/16042 fix this too?

maxwxyz, 2024-08-28:
@MisterMakerNL @kadet1090 ^

kadet1090, 2024-08-28:
@maxwxyz you are the one with MacOS - try it. It probably will for non-classic preincluded themes.

maxwxyz, 2024-09-03:
@kadet1090 still an issue on the latest weekly
```
OS: macOS 13.6.9
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.8.17
  * CfdOF 1.27.5
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.27
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.24
  * Curves 0.6.45
```

luzpaz, 2024-10-23:
@hyarion can you reproduce ?

hyarion, 2024-10-23:
Yes I can reproduce. This is one of the reasons for https://github.com/FreeCAD/FreeCAD/pull/17410#issuecomment-2431381751

adrianinsaval, 2024-11-13:
I don't know if we can truly consider this fixed, we rather just hid the option that led to this in the startpage

MisterMakerNL, 2024-11-13:
This could probably also be fixed by switching classic Qtstyle too fusion. But the point of classic is that it adapts too the OS Qtstyle. And so it kinda is absolete if we don't do that anymore.
Also fixing bugs in a stylesheet is waaaaay easier than doing it in classic.
So I'm very satisfied on how it is now.

maxwxyz, 2024-11-14:
Can someone test on mac + Qt6?

maxwxyz, 2024-11-15:
Removing the blocker tag as for 1.0 it was "fixed" with the workaround

closed 2025-07-15 by maxwxyz