FreeCAD/FreeCAD #15802: Stylesheet: Task dialogs have no header background and are not legible in Classic

maxwxyz, 2024-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With FreeCAD Classic, overlays and transparency enabled, the headers in the task dialog have no background and are not legible:
[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-08-06:
I did not touched overlay stylesheet in my last PR I have no clue what is going on here.

MisterMakerNL, 2024-08-14:
Still an issue?

maxwxyz, 2024-08-15:
I cannot switch to Classic to test

maxwxyz, 2024-08-24:
@MisterMakerNL yes, still an issue
```
OS: macOS 13.6.9
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

MisterMakerNL, 2024-08-24:
Might have a relation with https://github.com/FreeCAD/FreeCAD/issues/15799#issuecomment-2308427441
If we cross reference with Opentheme we can see if it's a stylesheet issue or an overlay issue in general...

maxwxyz, 2024-09-03:
@MisterMakerNL @kadet1090 still an issue in the latest weekly

```
OS: macOS 13.6.9
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.8.17
  * CfdOF 1.27.5
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.27
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.24
  * Curves 0.6.45

```

maxwxyz, 2024-09-03:
[image]

maxwxyz, 2025-07-15:
closing as classic is deprecated on Win/Mac.
@MisterMakerNL @kadet1090 is there a way to hide classic on these systems in the preferences?