FreeCAD/FreeCAD #15803: Warning for old backup format at startup

maxwxyz, 2024-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've labelled it as blocker so it won't get lost.

When the _old_ backup format .FCStd# is enabled, there is a warning on every start of FreeCAd, that this format is "deprecated as of 0.21". 

Is is now also deprecated in 0.22 / 1.0?
If the old backup format is really deprecated then
- remove the setting to use this old format
- remove the warning and check at startup

If the old backup format is still supported
- adjust the message to display the version number (ideally not hardcoded)
- the setting could be removed nonetheless IMO

[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-06:
@chennes do you have more insights into this?

chennes, 2024-08-06:
I believe we had planned to remove it, but never actually did so.

maxwxyz, 2024-08-06:
shall I leave it as 1.0 reminder?

hexvolt, 2024-08-08:
@maxwxyz @chennes (not sure who else) Guys, if you don't mind a comment from a bystander (I'm not a contributor to FreeCAD, but I work in the industry for 15+ years and have seen this issue in companies and organizations I worked for).

The issue I'm talking about is over time people naturally tend to overuse the `blocker` label as a means to draw attention to the issue, rather than designating a real blocker. And while it works on drawing attention, it ruins the standard workflows of software release cycle: after feature freeze and fixing major blocker bugs, a version gets released, then the minor bugs are addressed in minor releases.

I couldn't find a definition of a `blocker` label in the FreeCAD contribution guidelines, but I think it should be reserved for something that prevents using the application entirely or renders one of its main features unusable or malfunctioning (i.e. application crashes or produces errors when it shouldn't, or renders wrong results, etc). Like the truly blocking blocker.

I watch the FreeCAD issue tracker and I see what seem to be minor bugs sometimes marked as blockers. I.e. fixing a message on the start screen, or colors being not perfect in a certain combination of Dark/Light color scheme or some other corner cases that do not create functional limitations. While there is no doubt these issues need to be addressed, at the same time it is probably something that can be fixed later in a minor release. I've seen similar organizational issues in different software development teams when people start labeling `blocker` here and there, and then it just falls into never ending cycle of infinite bug fixing, and then some fixes introduce other bugs, and those bugs also need to be evaluated, and it just goes on and on and on. I'm following FreeCAD updates and what I start seeing is something very similar: 3 blockers fixed, 5 new added, then 5 blockers fixed, 10 new added, and when I check the blockers some of them do not seem like affecting functionality or posing significant limitations.

Anyways, just wanted to share my thoughts, take it as you wish, and thank you for your amazing work and for making FreeCAD better every day! I admire your work

Reqrefusion, 2024-08-08:
@hexvolt I think it's used for issues that need to be fixed before blocker 1.0. At least that's how I understand it. I recommend you to discuss it on discord or forums so that more people can participate.

closed 2024-09-02 by yorikvanhavre