FreeCAD/FreeCAD #15804: Qt error message on startup macOS

maxwxyz, 2024-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On every startup I get this error message, only on macOS:
[image]

@JohnOCFII could you confirm this?

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-08-07:
No message on:

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.92.0
  * OpenTheme 2024.7.30
  * fasteners 0.5.25
  * Curves 0.6.42

@maxwxyz are you using more than one monitor/display? I ask because:
https://ccp4serv7.rc-harwell.ac.uk/tracker/index.php?func=detail&aid=3045&group_id=21&atid=176

maxwxyz, 2024-08-07:
Yes, but FreeCAD launches on the MBP monitor.

depthoffocus, 2024-08-07:
I wonder if you would get the error message with that display turned off, regardless.

I don't think I have a way to test this myself as I have no proper second display, I just tried with an iPad as a Sidecar display but didn't see the error.

maxwxyz, 2024-08-07:
There is no error when I just use the MBP screen and the external monitor is not plugged in.
When FreeCAD is open and I plug it in, there is also no error, even when opening other files afterwards.

depthoffocus, 2024-08-07:
I also found this:

luzpaz, 2024-08-07:
@JohnOCFII do you mind confirming this ?

JohnOCFII, 2024-08-07:
> @JohnOCFII do you mind confirming this ?

It appears to only be an issue with an external monitor attached to the MacBook Pro.  I have a monitor on a shelf that I can hook up and try later today.

JohnOCFII, 2024-08-07:
Ok -- I attached a Sceptre 22" (diagonal) monitor via HDMI to my 2021 16" M1 MacBook Pro.  I restarted the Mac.  I then launched FreeCAD to my MacBook Pro built-in display.  No error messages.  I am on macOS 14.6 where as I noted Max on 13.6.7 in case that is relevant. 

[image]

```
OS: macOS 14.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.7.30
```

luzpaz, 2024-08-08:
Thanks JohnOCFII :+1:   
CC @ipatch do you have an external monitor you could test with ? TIA

ipatch, 2024-08-08:
unfortunately i do not have a way of testing this issue. 🤦‍♂️

goleon69, 2024-08-25:
Not sure helps, I downloaded the 38506 and tried to start with a Mac Pro from 2016, unfortunately it doesn't start as require the MacOs version that cannot be installed on this Mac .
[image]

luzpaz, 2024-08-25:
@oursland would you have a moment to reproduce ?

goleon69, 2024-08-25:
Yes, how can I help?

oursland, 2024-08-25:
@luzpaz I am unsure which issue you are asking to reproduce.  @goleon69's issue is unrelated to this bug.

I cannot reproduce @maxwxyz's message with multiple monitor's attached.

    OS: macOS 14.6.1
    Word size of FreeCAD: 64-bit
    Version: 0.22.0dev.38498 (Git)
    Build type: Release
    Branch: main
    Hash: 7e311ed4b96299c8b9909d86cda2410a53cbd498
    Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
    Locale: English/United States (en_US)
    Installed mods: 
      * OpenTheme 2024.8.10
      * A2plus 0.4.65
      * kicadStepUpMod 11.2.7
      * fasteners 0.5.26
      * freecad.cross 0.4.0
      * Assembly4 0.50.15

[image]

luzpaz, 2024-08-25:
@oursland apologies, yes I meant max's original ticket which you answered. It seems you can't reproduce.

oursland, 2024-08-25:
@goleon69 The macOS version issue is being tracked as https://github.com/FreeCAD/FreeCAD-Bundle/issues/294.

goleon69, 2024-08-25:
> tracked as FreeCAD/FreeCAD-Bundle#294

Thanks guys, indeed the problem is related to the opening with an older
version of the Mac OS. Unfortunately I have the old 0.22 de. That was
working  and now isn’t 🤔

maxwxyz, 2024-11-01:
Still have the issue on macOS 13.7.1 (latest I can get on Intel MBP). Closing this as no one can reproduce though.

closed 2024-11-01 by maxwxyz