FreeCAD/FreeCAD #15805: [Feature] Assembly: tangent constraint

Anthony-Gaudino, 2024-08-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I propose the addition of an Assembly feature that would allow to constraint elements using tangency.

This constraint would allow constraining faces and edges with tangency, imagine having two cones, then selecting one face of each one and applying this constraint would move the cones together and they would be constrained by the tangency, if one cone is grounded, then the other cone would be allowed to rotate around it, and move up/down. Same would happen if the face of two cylinders were selected.

### Full version info

```shell
OS: macOS 14.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git)
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Launcher
  * Nodes 0.1.36
  * InventorLoader 1.4.0
  * Assembly4.backup1719958896.85335 0.50.12 (Disabled)
  * Render.backup1719959316.120863 2024.6.5 (Disabled)
  * woodworking 0.21.2.33771
  * ThreadProfile.backup1719959316.6397839 1.90.0 (Disabled)
  * MeshRemodel.backup1719959315.871541 1.9.23 (Disabled)
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * Pyramids-and-Polyhedrons
  * Curves.backup1719958901.528519 0.6.36 (Disabled)
  * fasteners.backup1719958901.808955 0.5.20 (Disabled)
  * nurbs
  * sheetmetal.backup1719959316.3711638 0.4.21 (Disabled)
  * A2plus.backup1719958896.309429 0.4.64a (Disabled)
  * Manipulator 1.5.7
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * toSketch 1.0.1
  * Curves 0.6.42
  * Assembly4 0.50.15
  * Assembly3 0.12.2
```

### Subproject(s) affected?

Assembly

### Anything else?

There was already some talk about a similar feature on https://ondsel.com/blog/assembly-workbench-preview/

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-07:
> imagine having two cones

Regarding cones: #13958

But I'm not sure if Assembly has general-purpose tangent constraint like this: https://help.autodesk.com/view/INVNTOR/2024/ENU/?guid=GUID-FF270566-CA9E-420A-A00A-F52DD0FDFD8E

Anthony-Gaudino, 2024-08-07:
@maxwxyz I tried using the distance joint, but it didn't work for me.
When I select two cylinder or two cone faces what I expect is that the two bodies will be moved together and the tangency will be constrained, but this doesn't happen, if I move one of the bodies I can move it inside the other body, which in my view is not correct.

I also tried creating a cylinder with a simple flat Part face on it, but again, in this case, it didn't work well for me, first of all it doesn't rotate the face, so it goes across the cylinder, and I found it hard to rotate the face in the correct place. Even after rotating, it doesn't place the face on the round face of the cylinder. It works better if I rotate the face 90 degrees first and then I apply the constraint, but it still allows the plane to go inside the cylinder.

[assembly.FCStd.zip](https://github.com/user-attachments/files/16521244/assembly.FCStd.zip)

maxwxyz, 2024-08-07:
@PaddleStroke could you take a loot at this file please? It really seems to stuggle, maybe because the assembly is using a 2D plane for that constraint? When recreating I have trouble adding this face to the joint and when flipping the joint, FreeCAD crashes. Maybe it is because of the 2D element in the assembly?

PaddleStroke, 2024-08-07:
It is currently broken. It seems there was a problem with const char* being empty. The PR 15813 fix this, then distance of 0 between cylinder and between cylinder + plane are both working. Between cones it's not supported by the solver yet.

FEA-eng, 2024-08-07:
Can this issue be renamed to something like "Distance joint with zero distance not working properly"? Because the PR will close it but it won't implement a general-purpose tangent constraint as originally requested here, right?

PaddleStroke, 2024-08-07:
Yes, the bug was actually not relative to tangent, I think it was affecting most of the distance joints.