FreeCAD/FreeCAD #15814: On-view Parameters Buggy Behavior

OussamBrahim, 2024-08-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I noticed this issue when trying to create a circle, but it affects other tools as well:
if I move the mouse before typing a diameter (30), it works fine. But if I don't move the mouse and instantly type a number (30 again), the dimension is set to 30 radius instead, and I get an error
https://github.com/user-attachments/assets/991287e5-426d-44ba-96d3-3d6092601230
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-08-29:
Ok so I find that this bug happens only if you start a circle on another point/line, ie if there's an autoconstraint on the circle mid.

PaddleStroke, 2024-08-29:
I think this is a combination of 2 bugs.
- Bug 1 : typing 30 was enforcing a radius in all case. But this was never seen because this is the last step of this handler. So you type 30, it makes a circle of 60, then a diameter constraint is added shrinking back the circle to 30. Now the second bug prevents the creation of the diameter, making this first bug visible.
- Bug 2 : The autoconstraint on the mid point seems problematic.

I wonder if it is not linked to https://github.com/FreeCAD/FreeCAD/issues/15494

PaddleStroke, 2024-08-29:
Ok I think I see why bug2 is happening.
If you start the circle on a point (on the origin let's say) then you don't move the mouse, then you can see that there is a 'Point On Object' autoconstraint that is suggested. ie it wants to add an autoconstraint between the point used for the center and the circle curve. Which would result in a null circle. So it blows up.
[image]

closed 2024-08-31 by chennes