FreeCAD/FreeCAD #15815: Femmesh2mesh doesn't approximate 2nd order elements

goatchurchprime, 2024-08-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Gmsh can make second order elements used by some FEM packages that are quartic triangular splines.  The current implementation ( https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Fem/femmesh/femmesh2mesh.py ) of the FemMesh2Mesh feature ( https://wiki.freecad.org/FEM_FemMesh2Mesh ) ignores their shape.

I have a Macro [femoder2tomesh.zip](https://github.com/user-attachments/files/16530518/femoder2tomesh.zip) that can subdivide and plot them to a better tolerance, as seen in this image where the subdivision has been set to 4.

[image]

(Unfortunately it doesn't implement quad elements, as has recently been added: https://github.com/FreeCAD/FreeCAD/issues/15644 )

My Macro can also add an App::PropertyLink parameter to point back to the source FemMesh, as well as an App::PropertyIntegerList to state which mesh triangles came from which FemMesh elements.  This is necessary when the mesh is used for other manufacturing processes (eg carbon fibre filament winding) in order to map back any properties calculated on the Mesh triangles back onto the FemMesh for simulation.

This feature would require a pop-up window to set the options and the subdivision factor.  Since there is already a request for a pop-up window to set the deformation scale factor ( https://github.com/FreeCAD/FreeCAD/issues/11301 ), these features could all be combined.

### Full version info

```shell
v0.21.2 and most recent
```

### Subproject(s) affected?

FEM

### Anything else?

[femorder2torus.zip](https://github.com/user-attachments/files/16530506/femorder2torus.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-07:
Any help with FEM is welcome. Are you able to send a PR to implement your macro with a simple pop-up window or will you need some support with this?

goatchurchprime, 2024-08-07:
I can do, but it'll need some quality control first.  Where do I look for a an example of a feature that has a good quality pop-up that I can model this on?

FEA-eng, 2024-08-07:
> Where do I look for a an example of a feature that has a good quality pop-up that I can model this on?

As I've mentioned in that issue about missing scale factor in GUI, I would base this pop-up on the one of the [Part ShapeFromShape](https://wiki.freecad.org/Part_ShapeFromMesh) tool. That's the file: https://github.com/FreeCAD/FreeCAD/blob/74e5c3fafba7b0307c6b75e57ef648baae32f971/src/Mod/Part/Gui/ShapeFromMesh.ui