FreeCAD/FreeCAD #15816: Sketcher external geometry "argument not found"

Jnorichards, 2024-08-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was trying to use external geometry in a sketch, i.e. an edge from another sketch in the same document, which fails in  v0.22.0dev. Simplest steps to reproduce:
New file; Part Design WB, Create Body, Create Sketch on XY plane. Draw a 10 mm x 10 mm square, constrained with corner at the origin => fully constrained sketch. Close and pad to 10 mm to create a 10x10x10 cube.
Select a face of the cube, and create a new sketch on it. In the new sketch, create External Geometry from a cube edge. A line element is created. Close the sketch (drawing further to create a fully constrained sketch does not affect the problem).
Behaviour differs by version: 
[code]
OS: KDE neon 6.0 (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.4.27
  * Help 1.0.3
  * Assembly4 0.12.5
[/code]
Document model shows Valid: Internal Name: Sketch001 (i.e. as expected).
----
Same steps in 0.22.0dev (see version info below):
Document model shows 'argument not found, Internal name: Sketch001
Console reports "292.804 <App> Document.cpp(3152): exception in Unnamed#Sketch001 thrown: argument not found
Sketch001: argument not found"

Sorry if I have missed a duplicate of something; I am a very inexperienced user of FreeCAD, but discovered this while following a tutorial which works in a stable version.

### Full version info

```shell
[code]OS: KDE neon 6.0 (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.4.27
  * Help 1.0.3
  * Assembly4 0.12.5
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jnorichards, 2024-08-07:
I tried opening the file created and saved in 0.21.2 (working) with 0.22.0dev; the error behaviour is apparent as soon as Sketch001 is opened.

maxwxyz, 2024-08-08:
Could you share the zipped file here? 
Also, please try with the latest weekly, there have been recent fixes for external geometry: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Jnorichards, 2024-08-08:
I have just installed the dev version shown below, and can confirm that the external geometry works correctly, both for the 'steps to reproduce' trivial cube in a new file, and also works when opening a file created in 0.21.2 which uses an arc for external geometry.
For completeness, I attach a zipped FCStd which (a) fails in the way described using  dev.38001, and (b) succeeds in dev.38419.

[issue15816.FCStd.gz](https://github.com/user-attachments/files/16541989/issue15816.FCStd.gz)

I'd be happy to close this issue, unless I can help further - many thanks for the swift attention.
Jonathan

```
OS: KDE neon 6.0 (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git) AppImage
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * fasteners 0.4.27
  * Help 1.0.3
  * Assembly4 0.12.5
```

closed 2024-08-08 by maxwxyz