FreeCAD/FreeCAD #15817: Asembly's on-click LCS don't appear when link is inside a group

hasecilu, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Assembly WB have this handy feature which shows a LCS when you click on a Link that has been inserted on the assembly, the LCS lets you move and rotate the part.

The problem is that if you put the links inside a **Group** (inside the assembly) the LCS no longer appears, so you have to make more operations to transform the part. Constraints work fine.

[image]

### Full version info

```shell
OS: Arch Linux (Hyprland/hyprland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * ThreadProfile 1.90.0
  * FreeCAD-Movie 2024.2.15
  * freecad.gears 1.2.0
  * PieMenu 1.7.3
  * freecad_metal_workbench 0.0.1
  * Ship 2022.4.11
  * OpenTheme 2024.7.3
  * FreeGrid 2.0.0
  * FreeCAD-Reinforcement
  * Defeaturing 1.2.2
  * FreeCAD_SheetMetal 0.4.16
  * FreeCAD-Pyramids-and-Polyhedrons
  * yaml-workspace 0.1.2
  * woodworking 0.21.2.33771
  * Assembly4 0.50.10
  * FastenersWB 0.5.24
  * lattice2 1.0.0
  * Rocket 3.3.0
  * dodo 1.0.1
  * DidYouKnowThat
  * FreeCAD-motion
  * Profiles 0.0.1
  * Alternate_OpenSCAD 1.0.0
  * osh-autodoc-workbench 0.2.3
  * CfdOF 1.27.4
  * A2plus 0.4.64a
  * kicadStepUpMod 11.2.4
  * Nodes 0.1.36
  * frame 0.1.1
  * pcb 6.2023.1
  * FreeCAD_assembly3 0.12.2
  * Render 2024.7.4
```
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-08:
@PaddleStroke FYI

PaddleStroke, 2024-08-08:
Thank you for the report. I'll check this out

closed 2024-08-12 by chennes