FreeCAD/FreeCAD #15819: Active Std_Part loses Tree background color after Std_TransformManip

Roy-043, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a Std_Part.
2. In the Tree view the object has a background color, and its label is bold to indicate it is 'active'.
3. Right-click it in the Tree view.
4. Select "Transform".
5. Move the object and press the OK button.
6. Result: the Tree view background color is gone, but the label is still bold and the object is still 'active'.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-08:
Is this also happening in the stable version?
Could you attach screenshots?
@MisterMakerNL FYI

MisterMakerNL, 2024-08-08:
I don't think I am cable in fixing this. It's also not that big of a bug. As toggling the active body resets it. 
This is also in no-stylesheet.

Roy-043, 2024-08-09:
This issue also occurs in V0.21.
[image]
In the image from left to right: inactive part, active part, active part after transform

obelisk79, 2024-08-18:
This should probably have the stylesheet tag removed.

luzpaz, 2024-10-26:
@benj5378 any chance you could address this UI bug ?

maxwxyz, 2025-06-16:
@Roy-043 is this still an issue?

Roy-043, 2025-08-16:
Still an issue:
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42885 (Git)
Build date: 2025/08/08 15:11:51
Build type: Release
Branch: main
Hash: 0cdf0af0c6f9f4c12ab7072064906626f86f6f98
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/d1149ae28f1d2a91c1350b58814e279165517598

Leaving edit mode restores the saved tree background instead of clearing it, so an active Part keeps its highlight.