FreeCAD/FreeCAD #15821: "object":: 'App.Part' object has no attribute 'Shape'

ReneSci, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD quite often comes up with messages like
``` Connect_TotalAirMass: 'App.Part' object has no attribute 'Shape' ```
I can't figure out what causes this error and what to do

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: Danish/Denmark (da_DK) ]
Installed mods: 
  * A2plus 0.4.65
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * MeshRemodel 1.9.23
  * OpenDark 2024.3.13
  * OpticsWorkbench 1.0.18
  * QuickMeasure 2022.10.28
  * sheetmetal 0.4.20
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-08:
This seems to be related to a 3rd party addon/macro. Could you try this without any mods?

Roy-043, 2024-08-09:
A Std_Part object only has a Shape property if it contains visible objects with a Shape. Instead of making the content of the Part invisible, make the Part itself invisible.

luzpaz, 2024-08-23:
`Connect_TotalAirMass:`where does this come from ?

ReneSci, 2024-12-03:
Just rested the latest nightly build:
```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Danish/Denmark (da_DK) ]
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.65
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * MeshRemodel 1.9.23
  * OpenDark 2024.3.13
  * OpticsWorkbench 1.0.18
  * QuickMeasure 2022.10.28
  * sheetmetal 0.5.10
```
The following works as I would expect:
Fusion:
[image]

Cut:
[image]

Common:
[image]

Section:
[image]

However - these does not:
-----------------------------------------------------

Connect:
[image]

Boolean-Fractments:
[image]

Clice:
[image]

Compound:
[image]

maxwxyz, 2024-12-04:
@ReneSci I'm not sure how your last comment refers to the original issue?

ReneSci, 2024-12-04:
Sorry - wrong issue !
Should have been here:
https://github.com/FreeCAD/FreeCAD/issues/17718

closed 2025-07-30 by github-actions[bot]