FreeCAD/FreeCAD #15824: Selecting objects in groups inconsistent between tree and view

PaddleStroke, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Group
- Cube

Select the cube in the tree : 
`>>> # Gui.Selection.addSelection('unnamed','Box')`

Select the cube in the 3d view :
`>>> # Gui.Selection.addSelection('unnamed','Box','',48.4406,24.3605,7.5372)`

This is the same right? So far so good.

Now let's do : 

Part
- group
- - Cube

Select the cube in the tree : 
`>>> # Gui.Selection.addSelection('unnamed','Assembly','Group.Box.')`

Select the cube in the 3d view :
`>>> # Gui.Selection.addSelection('unnamed','Assembly','Box.',5.19714,0,1.89179)`

In this case Group does not appear in the subname.
@wwmayer 

- related to https://github.com/FreeCAD/FreeCAD/issues/15822
- Is causing : https://github.com/FreeCAD/FreeCAD/issues/15817

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-08-23:
Thanks for the feedback.
Can it be solved ? Or should I make a workaround to ignore groups in assembly?

wwmayer, 2024-08-24:
> Can it be solved?

If you modify the second line of the macro to be
`Gui.Selection.addSelection('unnamed','Part','Group.Box.',5.19714,0,1.89179)`
then it still works as expected.

So, all what should be adjusted is the selection code of the 3d viewer to not ignore the Group object as sub-element.

PaddleStroke, 2024-08-27:
> selection code of the 3d viewer to not ignore the Group object as sub-element

Where is this code located? I had a look before but couldn't figure it out. I'll check again

wwmayer, 2024-08-27:
As a start point try SoFCUnifiedSelection.cpp

PaddleStroke, 2024-08-28:
I found the root reason for https://github.com/FreeCAD/FreeCAD/issues/15822, but I cannot find for the DocumentGroupObject. There does not seem to be a getElementPicked in the ViewProvider of Groups so I'm not sure where this could come from.

As a side note, this issue is not causing a bug anymore as it was easy to fix the bug in assembly. But it would still be better to fix it.

PaddleStroke, 2024-12-04:
Bug still here.
@realthunder not a huge bug but perhaps you know why we have this inconsistency?

PaddleStroke, 2025-07-15:
Bug still here.

FEA-eng, 2025-11-15:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.20251111 (Git shallow)
Build date: 2025/11/11 16:45:10
Build type: Release
Branch: main
Hash: 4fbb389ce1295e04a5ad7f3a63f68b63b4f4036c
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```