FreeCAD/FreeCAD #15825: Part Cut Creates Geometry not present in original shapes

hhassey, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Doing  a boolean operation with two solids, subtracting one from the other, creates geometry not present in any of the original shapes

See details in the forum [HERE](https://forum.freecad.org/viewtopic.php?t=89404)

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git)
Build type: Unknown
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods:
* sheetmetal 0.4.18
* BOLTSFC 2022.11.5
```

### Subproject(s) affected?

Part

### Anything else?

This was [posted on the forum](https://forum.freecad.org/viewtopic.php?t=89404), and has no comments so far after 200+ views 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-08:
Could you please update the issues with the example file and screenshots and a step how to reproduce this or a recorded macro of the action.
I've downloaded the file but struggle to reproduce this as e.g. there isn't even a frames object so I'm not sure what you want to do.

hhassey, 2024-08-08:
I will as soon as I can. But the tread on the forum explains pretty well the process.

Try to cut the steel frames out of the "air" object. Using part cut.

-------- Mensaje original --------

hhassey, 2024-08-08:
@maxwxyz  here is the file, I have detailed the steps to reproduce above. 
Thank you for checking this out.

Don't forget to rename de file from Model2.forum.zip to Model2Forum.FCstd, github would not allow me to upload FreeCAD files. 

[Model2(forum).zip](https://github.com/user-attachments/files/16551267/Model2.forum.zip)

hhassey, 2024-08-08:
@maxwxyz Steps to reproduce are already numbered in the first post of this thread.  I am posting this because the Status says _Needs Steps to Reproduce_

maxwxyz, 2024-08-10:
Yes I can confirm. I thought maybe it is because of both the faces at the same position some OCC bug but even if I scale the air up or down I get a bad result. I've noticed that the Part_Scale command also behaves weirdly by moving or hiding the object (https://github.com/FreeCAD/FreeCAD/issues/15832)
@bgbsww is this TNP relevant or could you narrow this down?

bgbsww, 2024-08-24:
0.21.2 does the same thing.  Can't be TNP related.  Note that doing the intersection/common of the shapes also comes up with the geometry that we don't think belongs.

Note that there are a lot of Clone document objects in the Frame Fusion, and I'm suspicious of something involving them.

AIRCAP, 2024-10-09:
there might be a work around -- apply https://github.com/FreeCAD/FreeCAD/pull/17119  and in settings set boolean fuzzy value to 0,001 or lower (default of 1.0 can do the fusion correctly, but oddly enough not the cut) - no refinement needed

PR #17119 (2024-10-21): Fix boolean operations
Fix #17159
Fix #17150
Fix #17085
Fix #16081 
Fix #15825
Fix #15615
can alleviate #11083 ( fixes with nonstandard setting for boolean fuzzy > 1000 - but that's too high to cut out very delicate parts from very large blocks )
Fix #9666
Allows workaround for #17338 

OCCT has issues with boolean operations failing when parts are touching but not intersecting. This includes but is not limited to the PartDesign helix , which is notorious for producing results that do not pass BOP tests or even fail completely.

The suspected problem is too narrow precision. Part precision sho [... cut]