FreeCAD/FreeCAD #15827: fem: calculix: loading result failes

looooo, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

running a simple fem frequency analysis failed on a raspberry pi 5 due to failing to import the result. The error message is:
```
loading results failed
```

https://github.com/FreeCAD/FreeCAD/blob/8339bd9618c35331919d374d8f9abd92b79fdd46/src/Mod/Fem/femtaskpanels/task_solver_ccxtools.py#L289

it would be nice to get a traceback here to see what was wrong with the loading of the result.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (LXDE-pi-wayfire)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

looooo, 2024-08-08:
https://github.com/FreeCAD/FreeCAD/pull/15828

luzpaz, 2024-08-09:
@looooo does the PR fix this ticket ?

looooo, 2024-08-09:
yes, with this change the error was reported. It was a missing dependency (matplotlib). But I have no idea why it is used for loading the results of a frequency analysis with FEM. Seems like it is used somewhere else. maybe for plotting the histograms of the results.

commit 1a5ab14c (2024-08-12): FEM: print exception if loading a result in fem fails
Fixes #15827
closed 2024-08-12 by chennes