FreeCAD/FreeCAD #15830: Subtractive Loft is activated at the same time with the previous Additive Loft.

leoheck, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Subtractive Loft is activated with Additive Loft.

[image]

If I toggle its visibility, the issue will be fixed.
[image]

However, if I double-click to edit the Subtractive Loft, it enables the Addictive Loft (or any feature behind it)

Steps to reproduce:

1. Create a Body with PD WB
2. Create some sketches (I am using 4), stack them vertically
3. Create the Additive Loft.
4. Copy the Sketches, and place inside the Body again.
5. Shrunk them a bit
6. Apply the Subtractive loft. 

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git) AppImage
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.16-dev
  * sheetmetal 0.4.23
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.30
  * fasteners 0.5.25
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.19
  * lattice2 1.0.0
  * kicadStepUpMod
  * OpticsWorkbench.backup1722883583.3368473 1.0.18 (Disabled)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-08:
Could you share the file and even better: record a macro how you reproduce this?

leoheck, 2024-08-08:
> Steps to reproduce:
Create a Body with PD WB
Create some sketches (I am using 4), stack them vertically
Create the Additive Loft.
Copy the Sketches, and place them inside the Body again.
Shrunk them a bit
Apply the Subtractive loft.

leoheck, 2024-08-08:
Double-click the SubtractiveLoft001 here, and it will enable the Additive Loft. However, in this example, the Additive Loft does keep active after hitting Ok to close the edition.
[Unnamed1.zip](https://github.com/user-attachments/files/16552393/Unnamed1.zip)

maxwxyz, 2024-08-10:
I can reproduce on activation but not sure if this is an intended behavior, like when selecting edges to fillet. When select mode is enabled, the previous feature is also visible. Not sure if this is the desired behavior for a subtractive loft @pierreporte what do you think?
I **cannot** reproduce that it stays active after closing the subtractive loft dialog. Regardless of clicking OK or Cancel, the dialog closes and the additive loft is hidden again, only the last feature remains active.
@bgbsww as always :)

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

pierreporte, 2024-08-10:
I cannot reproduce the additive loft remaining active after closing the subtractive loft.

Besides, the current behavior seems normal to me. When a feature is edited, the tip is moved to the immediate previous feature. Other CAD programs work like this.

leoheck, 2024-08-10:
> When a feature is edited, the tip is moved to the immediate previous feature. Other CAD programs work like this.

I see. I don't know other CADs, however the behavior is different than the other features of the PDWB.

maxwxyz, 2024-08-11:
I'll close this then if works as designed.

closed 2024-08-11 by maxwxyz

pierreporte, 2024-08-11:
@leoheck Now I realise that indeed other features work differently. The subtractive loft is actually showing a preview of the feature while other can’t. Feature preview was requested in #11214.

leoheck, 2024-08-11:
@maxwxyz it doesn't work as designed. see the fist message on this thread. The first image shows both features active after the edition has closed.