FreeCAD/FreeCAD #15831: Stylesheet: Classic has dark background in task dialog

maxwxyz, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With FreeCAD Classic some element selection boxes have a dark background in the task dialog, compared to the other light shades. Here an example of Part Scale:
[image]

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-08-14:
is fixed

closed 2024-08-14 by maxwxyz

luzpaz, 2024-08-15:
@MisterMakerNL if possible, would you link to the PR or commit that provides the fix ? Or you can even auto-remotely close tickets by referencing them in your commit messages https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue  
(Apologies if you already know about this feature)