FreeCAD/FreeCAD #15832: Part Scale hides part on default / moves object

maxwxyz, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Not sure if I'm doing something wrong or if if there is a bug in the Part Scale tool from: https://github.com/FreeCAD/FreeCAD/pull/10583

I'm trying to recreate a bug with the file from: https://github.com/FreeCAD/FreeCAD/issues/15825
Every time when I apply this tool, the object disappears / hides and I need to expand the Part Scale object and change the visibility of it's child to see it in the 3D view. When applying the tool again on the second part or adjusting the scale values of the first operation in the properties view it moves the scaled copy and displays it twice now.

Here is the demo video:

https://github.com/user-attachments/assets/bf21d3d8-e954-4789-ae48-e89c99f568a7

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.30
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.25
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.23
  * Curves 0.6.42
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-08:
@WandererFan FYI, you've made the PR that introduced the tool in the Part WB

maxwxyz, 2024-08-10:
@bgbsww maybe you have any idea about this, could this be related to TNP fixes or could it be related to https://github.com/FreeCAD/FreeCAD/issues/14517 
@Roy-043 FYI

bgbsww, 2024-08-11:
I don't think it's related to either.

maxwxyz, 2024-08-11:
@wwmayer any idea about that behavior?

WandererFan, 2024-08-17:
Hiding the input object is standard behaviour in the Part WB.   

As for the moving, the scale transform needs a point as "the center of the scale".  Current it uses the origin, so if your part is symmetric around the origin, the scaled object won't move about.  If the part is not on the origin, it will be mirrored around the origin.  (just learned this today!).

I can pick a different point as the center.  The choices would be to use the object's placement or the geometric center of the object's shape.  I lean towards the geo center.  Any thoughts?

maxwxyz, 2024-12-07:
Closing as intended behavior

closed 2024-12-07 by maxwxyz