FreeCAD/FreeCAD #15833: About FreeCAD does not distinguish between ARM and X-86 compiles.

edwilliams16, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

About FreeCAD does not distinguish between Mac ARM and X-86 compiles.  This would be needed for debugging.  I also can't verify which version I'm actually running in a given instance.

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.23
  * DynamicData 2.62.0
  * OpenTheme 2024.6.22
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal 0.4.20
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.40
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-09:
@adrianinsaval should this be in this repo or the bundle one?

adrianinsaval, 2024-08-09:
> @adrianinsaval should this be in this repo or the bundle one?

no it should be here, probably we should change `Word size of FreeCAD: 64-bit` to processor architecture instead, virtually nobody is running 32bit devices anymore so that differentiation is useless.

JohnOCFII, 2024-08-09:
> it should be here since it would need a change in source code, probably we should change `Word size of FreeCAD: 64-bit` to processor architecture instead, virtually nobody is running 32bit devices anymore so that differentiation is useless.

That makes sense.  Also, it won't be long before there are more requests for FreeCAD on Arm for Windows too. https://spectrum.ieee.org/windows-on-arm

PR #17541 (2024-10-29): Gui: show cpu architecture in version info instead of word size
Fix #15833
commit 9ab3bef6 (2024-10-29): Gui: show cpu architecture in version info instead of word size

Fix #15833
commit 065029b6 (2024-10-29): Gui: show cpu architecture in version info instead of word size

Fix #15833
closed 2024-10-29 by chennes