FreeCAD/FreeCAD #15834: [Problem] Sketcher: Rounded corners frame rectangle with negative thickness crashes FreeCAD

MiqCG, 2024-08-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Sketcher, when drawing a "Centered rectangle" with modes "Rounded corners" and "Frame" activated, and On-View-Parameters (OVP) "Dimensions only", FreeCAD crashes when entering the final number for the thickness with a negative value (decimal numeral or not). See attached video.

With commercial tubes you normally know the external dimensions and the thickness.
The result is correct if "Rounded corners" are deactivated (only Frame mode).

Tested and found in 0.22 revisions 38419, 38385, 37543 and 35966. I suspect that somebody has spotted this problem before but I haven't found any related issue.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git) AppImage
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * sheetmetal 0.4.21
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * fasteners 0.5.24
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/user-attachments/assets/e44265bc-7033-4763-8900-eac59c9bfce4

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-09:
As usual, no crash but errors on Windows:

```
11:56:57  Unnamed#Sketch: Failed to add box
11:56:57  Tool execution aborted
11:56:57  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

maxwxyz, 2024-08-09:
@PaddleStroke FYI

PaddleStroke, 2024-08-29:
It appears that it bugs only if the radius and the negativ internal thickness are equal.
So radius = 2, thickness = -2 for example.

PaddleStroke, 2024-08-29:
Ok I see why.
When you use a negativ thickness, there are 2 cases possible. Either thickness is more than radius, then the inner rectangle is a rectangle. Either it's less than radius, then it has round corners.
[image]

The case where thickness = radius is problematic because you either have null arcs. Or the construction lines that are null.
So this case needs to be handled specifically.

closed 2024-09-09 by yorikvanhavre