FreeCAD/FreeCAD #15837: TechDraw: Broken View can't use a sketch from inside a Body

FEA-eng, 2024-08-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The sketch defining the break lines for a Broken View has to lie outside of a Body. Otherwise, there's an error:

`TechDraw did not retrieve any geometry for View/View`

I'm not sure if it's a limitation or a bug but looks more like the latter for me.

[Broken view sketch.zip](https://github.com/user-attachments/files/16559647/Broken.view.sketch.zip)

Forum discussion: https://forum.freecad.org/viewtopic.php?t=89641

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-08-09:
Confirmed. Note that you can drop the Sketch into the Body once the Broken View is created.

FEA-eng, 2024-08-09:
> Note that you can drop the Sketch into the Body once the Broken View is created.

Right, it allows modifications then. So it seems to be a bug if it doesn't work the first time.

maxwxyz, 2024-08-09:
@WandererFan FYI

WandererFan, 2024-08-17:
If the sketch is inside a body, it is not treated as an object by Selection, but arrives as Body + subelement(sketch).
When we check through the list of selected objects for a sketch we don't see it.  

Need to add code that hunts for subelements that might be the break object.