FreeCAD/FreeCAD #15842: "link transform" of Draft link array is ignored if the array is used in operations of other workbenches

LupinIII, 2024-08-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A Link array with "Link Transform" set to true shows the linked objects in the right position (honoring the transform of the original objects), but any operation of the Part workbench on the link array will have the result in the wrong position (as if "Link Transform" wasn't set).

Pretty much all operations seem to ignore it. Booleans, "Mirror", "Fillet", even just putting the link array into a compound will make an object as if the transform wasn't there.

The same is true if the link array is used as a base feature of "Part Design" body, a "Draft" clone or Shape2DView (and very likely more).

See the attached file for an example.
This is possibly related to #15378 

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * CAM
  * Draft
  * Fem
  * Help
  * Idf
  * Import
  * Inspection
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot
  * Points
  * ReverseEngineering
  * Robot
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TabBar
  * TechDraw
  * Test
  * Tux
  * Web
```

### Subproject(s) affected?

Draft

### Anything else?

[link_transform_ignored.zip](https://github.com/user-attachments/files/16566629/link_transform_ignored.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-29:
@Roy-043 is this also a Core related issue instead of Draft ? (as you refer to in https://github.com/FreeCAD/FreeCAD/issues/15378#issuecomment-2248317411)

Roy-043, 2024-08-31:
I was not aware of this behavior. I would be inclined to use "Extra Placement" property of the array to achieve this result. But reading through the documentation this behavior seems to be intended.

> is this also a Core related issue

Yes, this is Core related. The Draft Link arrays rely on code from Core. Note that in this case the Shape of the array is not correct (as opposed to #15378).