FreeCAD/FreeCAD #15846: Python very slow

Jolbas, 2024-08-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Running python has been extremely slow in FreeCAD 0.22 compared to 0.21. Tested on several builds according to table below. The problem is not limited to the attached simple performance test but it seems to depend on if script is mainly FreeCAD functionality or mainly pure python.
For example there is no difference for Test_Testbase script in Test Framework:
0.21.2: Ran 44 tests in 6.41s
0.22-38429: Ran 48 tests in 6.43s

But for this pure python script the difference is huge.

```python
from time import perf_counter
def simple_speed_test():
    t = perf_counter()
    r = 3.9
    x = 0.5
    for _ in range(1_000_000):
        x = r*x*(1-x)
    print(perf_counter() - t)

simple_speed_test()
```

Results on my system:
In MacOS Terminal
python 3.10.12: 0.05s
python 3.12.3: 0.06s

In FreeCAD python interpreter
build 33694: 0.07s (0.21.1 python3.10.12)
build 33771: 0.07s (0.21.2 python3.10.13)
build 34594: 10.4s (0.22 python 3.10)
build 38429: 10.5s (0.22 python 3.11)

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-08-10:
Confirmed, looks like I've narrowed it down to approx 70 commits somewhere in mid April this year.

Last build working as expected:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36807 (Git) AppImage
Build type: Release
Branch: main
Hash: cc96f2718e139d670e08c4226bdea9955a102ed6
Python 3.11.8, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * Curves 0.6.42
  * CfdOF 1.27.4
  * fasteners 0.5.25
  * sheetmetal 0.4.23
```

first build running very slowly:

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36874 (Git)
Build type: Release
Branch: main
Hash: a4dd600ec6d5a53a9960cd9f8693d7d307b9e359
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.3.0
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * Curves 0.6.42
  * CfdOF 1.27.4
  * fasteners 0.5.25
  * sheetmetal 0.4.23
```

bdieterm, 2024-08-10:
Can also confirm.
Looks like tracing is activated which slows down the execution. (Tracing is normally used for breakpoints etc..)
Inserting the line `import sys; sys.settrace(None)` at the beginning of the code snipped brings back normal execution speed.

Syres916, 2024-08-10:
So the reason my AppImage build 36807 was working fine was because it was using my 0.21 config files which didn't have that parameter in them, darn.

Syres916, 2024-08-10:
Maybe the Wiki entry could be enhanced to give the user a better understanding of the consequences of this setting, so changing the text in the table of https://wiki.freecad.org/Preferences_Editor#General_3 from:

`The interval at which the profiler runs when there is Python code running (to keep the GUI responding). Set to 0 to disable. `

to something along the lines of:

```
The interval at which the profiler runs when there is Python code running (to keep the GUI responding). Set to 0 to disable.
Note: this setting has a significant performance impact on non-Gui Python scripts, set to 0 if existing code has slowed down since upgrading but beware the probability of crashes increases when running Gui scripts.
```

@Roy-043 by all means chime in if you have a more suitable set of words for this.

bdieterm, 2024-08-10:
@Syres916:
I don't know yet what the exact reason for slowing down the execution is. I just documented my observations here.

> So the reason my AppImage build 36807 was working fine was because it was using my 0.21 config files which didn't have that parameter in them, darn.

This is strange. Could you check that again? I would have expected that the default value for ProfilerInterval (which is 200 ms) would have been used for an older config that lacks that parameter.

Syres916, 2024-08-10:
I agree with the expected default value, hence I double checked before making the comment. When I get back to my PC I'll post the _PythonConsole_ parameters exactly how they are in my 0.21 config.

Syres916, 2024-08-10:
In order to give complete info, here is the _user.cfg_, _system.cfg_ and _FreeCAD.conf_ for 0.21 that when using 0.22dev runs the Python script above in just over seven hundredths of a second:

[FreeCAD021configs.zip](https://github.com/user-attachments/files/16570369/FreeCAD021configs.zip)

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38394 (Git)
Build type: Release
Branch: main
Hash: 6fda739a14b57da0c7d84230c17933738493d5bf
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * Curves 0.6.42
  * CfdOF 1.27.4
  * fasteners 0.5.25
  * sheetmetal 0.4.23

```

bdieterm, 2024-08-10:
I tried the old 0.21 configs provided by @Syres916 with the current 0.22 AppImage, but this had no positive impact on the performance.

Other things I tried:
Using the current AppImage 0.22.0dev.38429 I get around 1.5 s vs. 118 s execution time with disabled resp. enabled tracing, and using 10e6 cycles.
But with the self-compiled main branch in a docker image, the performance difference is much smaller: 2.5 s vs. 8.5 s.
As far as I have seen, the code inside the execution tracing callback consumes most of the extra time. Immediately returning inside the [tracing callback ](https://github.com/FreeCAD/FreeCAD/blob/aab0b31aec8d30691c3b012023e7a26fcc627b2a/src/Gui/PythonTracing.cpp#L128) reduces the execution time from 8.5 s to 3.9 s.
Since the callback is called for every Python opcode execution, I tried to optimize that -- see #15855. This reduced the execution time to around 4.0 s.

I don't know if this will fix the much bigger performance difference that I could only see when using the AppImage, so let's keep this issue open.

Syres916, 2024-08-11:
> I tried the old 0.21 configs provided by @Syres916 with the current 0.22 AppImage, but this had no positive impact on the performance.
> 

Astounded is the only word I can come up with at this late hour!! I get `0.06941013599862345` seconds using the same AppImage on a 12 year old system with those config files.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git) AppImage
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * Curves 0.6.42
  * CfdOF 1.27.4
  * fasteners 0.5.25
  * sheetmetal 0.4.23

```

bdieterm, 2024-08-11:
@Syres916
> I get 0.06941013599862345 seconds using the same AppImage on a 12 year old system with those config files.

I used a slightly different code for my performance tests before (10e6 cycles instead of 1e6, and with the code outside a function which makes it slower) because the performance-difference (with/without tracing) was not so large on my self-compiled FreeCAD.

To have a better comparison, I re-ran the tests with your specifications:
* FreeCAD_weekly-builds-38429-conda-Linux-x86_64-py311.AppImage
* your three FC 0.21 config files in my `~/.config/FreeCAD` directory, replacing my original ones --> I see your different macro path and different colour scheme after starting FreeCAD.
* a very old, low-spec computer, running Linux Mint 20.2
* the benchmark code posted by @Jolbas

Starting FreeCAD, then running the benchmark code as a macro gives the following output:
10.991051381000034

But if I change parameter "ProfilerInterval" to 0 or insert line `import sys; sys.settrace(None)` at the beginning of the code, then I get the output of around 0.083 s.

Theses execution times are very similar to what @Jolbas measured on a macOS computer.

But it is really strange that I have to switch off tracing to get similar execution times (around 0.08 s) as you.

Roy-043, 2024-08-12:
I am a bit confused, but this may be over my head anyway.

In the commit https://github.com/FreeCAD/FreeCAD/commit/e78f74b01a44bce0853a847252ec5f33294e1c36 I read:

> Made the Python profiler only run when the console is running code

> Python from the console now runs in the background

If Python runs in the background it should not make the GUI unresponsive. Why is ProfilerInterval still relevant then?

bdieterm, 2024-08-12:
> Python from the console now runs in the background

But there is also a commit message
> Removed the background thread running Python code and replaced it with a custom
profiler which the Python interpreter runs frequently ... on the main thread whenever Python code is running.

I don't know any details about that commit, so take this with a grain of salt.

ProfilerInterval makes a difference if you run code such as
```python3
import time
t_start = time.time()
while time.time() < t_start + 15.0:
    pass
```
as a macro, or in FreeCAD's Python console widget.

When set to zero, the GUI freezes during these 15 seconds. Otherwise the GUI will still respond because the Qt events are processed regularly.

To be honest, I am not excited about that "prevent gui freezing while a macro is executed" feature. I prefer the old behavior which freezes the GUI while the macro is running. This makes it easy to know when the execution has finished and when FreeCAD is ready to process new (Python) commands.

Roy-043, 2024-08-12:
@bdieterm Thanks for clarifying.

@Syres916 Your text mentions the 'probability of crashes':

> The interval at which the profiler runs when there is Python code running (to keep the GUI responding). Set to 0 to disable.
Note: this setting has a significant performance impact on non-Gui Python scripts, set to 0 if existing code has slowed down since upgrading but beware the probability of crashes increases when running Gui scripts.

Is there really that much reason for concern?

Roy-043, 2024-08-25:
What is very confusing for me is that after entering a line of code that takes a long time, the cursor keeps blinking at the end of the line as if Enter has never been pressed. My 'automatic' response is to press Enter again. I have now seen the "Previous command still running!" warning many times...

benj5378, 2024-09-01:
> I don't know yet what the exact reason for slowing down the execution is. I just documented my observations here. @bdieterm 

It is because of the profiler, as can be seen on this Valgrind-KCachegrind. However, within the profiler, 99% of time is consumed by `QTime::currentTime()`, 85% within FreeCAD. This does not seem right. Let's have a look at the logic.

First, the trace is set to `tracer_callback`

https://github.com/FreeCAD/FreeCAD/blob/2a9cf9e8e6e62086948eb15e24d1e69a54f53d97/src/Gui/PythonTracing.cpp#L108-L121

In `tracer_callback`, we check the time since last Qt update by using `QTime::currentTime()`.

https://github.com/FreeCAD/FreeCAD/blob/2a9cf9e8e6e62086948eb15e24d1e69a54f53d97/src/Gui/PythonTracing.cpp#L123-L152

 If more than 200 ms (or what is set in preferences) has passed, then we call for a GUI update. The 85% of time is used on checking whether 200 ms has passed. While I ran Valgrind, `tracer_callback` was called checking time 8.098.232 times. To comparison PyNumer_Multiply was called 8.098.232 times. That basically means, that for each iteration of
```
    for _ in range(1_000_000):
        x = r * x * (1 - x)
```
`tracer_callback` (and QTime::currentTime) is called twice. Insane. However, according to [Python documentation](https://docs.python.org/3/library/sys.html#sys.settrace), _The trace function is invoked (with event set to 'call') whenever a new local scope is entered_.

So is this an edge case of misuse of FreeCAD Python console? Whenever would we anyone make code with so many scopes? They probably wouldn't. But maybe running Python modules within FreeCAD from the console, will do a similar number of scope entering/trace triggers?

Maybe this could be fixed using `sys.setswitchinterval` instead? [Python documentation](https://docs.python.org/3/library/sys.html#sys.setswitchinterval): _Set the interpreter’s thread switch interval (in seconds). This floating-point value determines the ideal duration of the “timeslices” allocated to concurrently running Python threads_. This might allow for calling `tracer_callback` with a fixed time interval; instead of calling it every few lines of code with `sys.settrace`. However, maybe the Python interpreter should just have it's own thread?

> Python from the console now runs in the background

Because it actually doesn't run in the background. It runs in the foreground on the same thread, just constantly being interrupted by interval checking for Qt update. Moving the Python console to an actual separate thread is the optimal solution, in my opinion.

What do you guys think?

[image]

benj5378, 2024-09-01:
> What is very confusing for me is that after entering a line of code that takes a long time, the cursor keeps blinking at the end of the line as if Enter has never been pressed. My 'automatic' [...]

I agree, I was also pretty confused. I think we should do a separate issue for this. If we can't run another command anyway (_Previous command still running!_), then we should just lock the console anyway, so users know what is going on.

Jolbas, 2025-01-19:
I think this is fixed by #15855 and can be closed. I don't have this problem any more and the test script runs in 0.07s with build 39900

benj5378, 2025-03-24:
@maxwxyz please close this issue

closed 2025-03-24 by maxwxyz