FreeCAD/FreeCAD #15847: Additive Loft Not Created in 0.22.0dev

ShayGus, 2024-08-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When adding additive loft from one sketch to another the result is one of two:
1. AdditiveLoft: BRep_API: command not done
2. AdditiveLoft: Loft: At least one section is needed
In either result, the loft is not created or created in an error state.
The same file works in 0.21.2

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Israel (en_IL)
Installed mods: 
  * Curves 0.6.42
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-10:
Could you please upload the demo file and even better, also record a macro of the steps you do and stop recording after the error?

maxwxyz, 2024-08-10:
@bgbsww FYI

ShayGus, 2024-08-11:
`# -*- coding: utf-8 -*-

+++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import PartDesign

# Gui.runCommand('Std_DlgMacroRecord',0)
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('additive_loft_demo','Body','Sketch002.')
### Begin command PartDesign_AdditiveLoft
App.getDocument('additive_loft_demo').getObject('Body').newObject('PartDesign::AdditiveLoft','AdditiveLoft')
App.getDocument('additive_loft_demo').getObject('AdditiveLoft').Profile = App.getDocument('additive_loft_demo').getObject('Sketch002')
App.ActiveDocument.recompute()
App.getDocument('additive_loft_demo').getObject('Sketch002').Visibility = False
App.ActiveDocument.recompute()
# App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.ShapeAppearance=getattr(App.getDocument('additive_loft_demo').getObject('Revolution').getLinkedObject(True).ViewObject,'ShapeAppearance',App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.ShapeAppearance)
# App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.LineColor=getattr(App.getDocument('additive_loft_demo').getObject('Revolution').getLinkedObject(True).ViewObject,'LineColor',App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.LineColor)
# App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.PointColor=getattr(App.getDocument('additive_loft_demo').getObject('Revolution').getLinkedObject(True).ViewObject,'PointColor',App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.PointColor)
# App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.Transparency=getattr(App.getDocument('additive_loft_demo').getObject('Revolution').getLinkedObject(True).ViewObject,'Transparency',App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.Transparency)
# App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.DisplayMode=getattr(App.getDocument('additive_loft_demo').getObject('Revolution').getLinkedObject(True).ViewObject,'DisplayMode',App.getDocument('additive_loft_demo').getObject('AdditiveLoft').ViewObject.DisplayMode)
# Gui.getDocument('additive_loft_demo').setEdit(App.getDocument('additive_loft_demo').getObject('Body'), 0, 'AdditiveLoft.')
# Gui.Selection.clearSelection()
### End command PartDesign_AdditiveLoft
# Gui.Selection.clearSelection()
# Gui.Selection.addSelection('additive_loft_demo','Body','Sketch003.Edge4',-14.6969,-0.877704,80)
# Gui.Selection.clearSelection()
App.getDocument('additive_loft_demo').recompute()
# Macro End: C:\Users\sgusi\AppData\Roaming\FreeCAD\Macro\Additive Loft Failure.FCMacro +++++++++++++++++++++++++++++++++++++++++++++++++`

[additive-loft-demo.FCStd.gz](https://github.com/user-attachments/files/16571731/additive-loft-demo.FCStd.gz)

bgbsww, 2024-08-11:
First, thanks so much for including the macro.  It makes it easy to be sure we're trying exactly the same thing you did.

What is glaring in that macro is that the AdditiveLoft.Sections property is not being set, and so the "At least one section required" error is valid.  When I manually set the Section to "TopProfile", I get the below, which I assume is right.

So this looks to me like some UI error that isn't setting the sections when the second sketch has that parametric position constraint.  That's worth somebody looking into.

Is this what we expect to see?
[image]

ShayGus, 2024-08-11:
This is what expected

maxwxyz, 2024-08-12:
@kadet1090 FYI for potential UI dialog code error

AIRCAP, 2024-09-24:
See #16630  --  they are actually the same thing - apparently the Task dialog selects indivudual edges/vertices instead of the whole part/sketch one has clicked on in 0.22 and later - and theres no way to select the whole sketch or part once a single edge/vertex has been clicked on -- previously clicking "add" and then clicking on something added the whole thing.

AIRCAP, 2024-09-24:
It looks like #16791 fixes this - can you confirm?  

regarding the macro code above - it looks like the GUI action of "Add Section" and "Remove Section" don't add respective macro code, so that is missing from the above macro -- but I think that's not the issue - the issue I think was that selecting the "Edge4" did not make the loft.   this is fixed with  #16791

bgbsww, 2024-09-24:
OT:  Macros:  Correct - generally none of the Gui actions in Macros affect the actual document itself.  Correctly placing the python commands in the Macro (and the console) is required to have been written by the author of the particular operation.  See https://github.com/FreeCAD/FreeCAD/pull/16671 for a recent example of a complete python workbench rework; there is similar work required for c based workbenches or operations that don't update Macros.

luzpaz, 2024-09-25:
Closing as fixed by #16791 
If reproducible in weekly please re-open.

closed 2024-09-25 by luzpaz