FreeCAD/FreeCAD #15848: [TechDraw] Exported PDFs are improperly generated

pure-machinery, 2024-08-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When exporting a TechDraw page to a PDF (via File -> Export) and later opening said PDF the drawing template is offset by an obscure amount. The geometry does not seem to be offset at least at a glance. 

This has been happening for me in the previous development versions as well. Aside from that linux is also affected.

I've added the test files and also a quick picture.

[image]

 
[td_template_bug.zip](https://github.com/user-attachments/files/16569858/td_template_bug.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenDark 2023.12.17
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-10:
How do you export the TechDraw page? You've mentioned File>Export but what did you choose there? Technical drawing or PDF? Please try with Technical drawing and type .pdf in the filename.

pure-machinery, 2024-08-10:
My mistake, sorry. I've choosen `File -> Export PDF`; the original post is also edited.

The way you recommend also does not work for me.

maxwxyz, 2024-08-11:
I can reproduce this issue in the dev version with filetype PDF. It looks normal when exporting as SVG.
I cannot reproduce this issue in the stable 0.21.2 version, so it seems like a regression in the PDF export.
@WandererFan FYI

commit df645583 (2024-09-02): [TD]fix page margins on pdf export (fix #15848)
closed 2024-09-02 by yorikvanhavre