FreeCAD/FreeCAD #15850: Discrete degrees of freedom not counted – under-determined sketches are reported to have 0 degrees of freedom.

keinstein, 2024-08-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD doesn't consider descrete degrees of freedom when it counts degrees of freedom. This leads to lots of unexpected results. Consider the following image:
[image]

Let's first understand why FreeCAD claims that the sketch is fully constrained:
- the modulus of each coordinate is given for the corner with the smallest distance to the coordinate origin by the vertical and horizontal lines ending in the axis.
- both points are constrained to be on the same side of the y axis as the vertical lines end in the same point.
- One line of the triangle is constrained to be horizontal and of 50 mm length.
- One line of the triangle is constrained to the length of 30 mm
- one angle of the triangle is given to be 30°

The image is reported to be fully constrained. However it has at many degrees of freedom:
- each of the triangles can have either shape which makes one degree of freedom for each of them (2 in total)
- each of the triangles can be mirrored on the vertical and/or horizontal axis across its anchor point (+4 degrees of freedom)
- The entire structure can be mirrored with respect to the Y axis (+1 degree of freedom)
So we have at least 7 degrees of freedom leading to at least 128 solutions. 

The solver does a relatively good job as it seems to look for the nearest local solution. However, editing such sketches can be a pain: Changing some size may completely mess up the drawing. And it must be sorted out by hand: Disable the constraints, move all vertices to the right positions, enable the constraints again, and hope that nothing bad happens. Otherwise you have to start again. Each enabled constraint triggers a refinement of the coordinates which may lead to bad starting conditions for the next constraint to be enabled. So everything has to be revised after each enabled constraint.

Currently I don't see any tools to solve this problem in FreeCAD. It occurs naturally in many situations because we measure distances. I first came across it when I wanted to sketch my new flat two years ago. I don't know the angles in my flat. But I can be quite sure that none of them is a right one. And the only way to measure the flat is by measuring distances, which cannot be treated as coordinates directly. 

I'd like to suggest an improvement: Currently a system of equations of continuous variables (distances, coordinates and similar numbers) seems to be solved. Adding the concept of orientation can help to reduce the maintenance effort in sketches. In a plain, the orientation is typically modelled by one of its normal vectors, while only its direction information is evaluated. For example given a triangle with coordinate vectors $a,b,c$ of its corners, the three vectors $(b-a)\times (c-b)$, $(c-b)\times (a-c)$ and $(a - c) \times (b-a)$ are all parallel pointing into the same direction, while the arrow $(b-c)\times (a-b)$ points into the opposite direction. This can be used to model the constraint “keep this point on the same side of this line”. If represented in the sketch-inherent 2D coordinate system, the orientation can be represented by a scalar that must not change sign during recomputation of all coordinates. This can be used to construct orientation constraints for larger polygons and convexity  constraints.

Such orientation can still be fooled if two points defining a line are exchanged during recalculation. So it should be sufficiantly easy to change the sign of an orientation constraint. This should also be possible if it leads to an unsolvable state, as sometimes it is necessary to cross invalid configurations in order to get a valid one.

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.
Build type: Release
Python 3.12.4, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[descrete Underconstrain.zip](https://github.com/user-attachments/files/16570080/descrete.Underconstrain.zip)

A sketch proving the problem.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-08-10:
Good old flipping ). 
The task is no worse than TNP

maxwxyz, 2024-08-11:
Could be the same as https://github.com/FreeCAD/FreeCAD/issues/8324 
If so, please add the issue as comment there.

keinstein, 2024-08-12:
@maxwxyz 
> Could be the same as #8324 If so, please add the issue as comment there.

I disagree. In #8324 the solver somehow has the information that the drawing is not fully constraint. The lower help line is dark blue meaning its not fully constraint. Even a numerical solver should come to the conclusion that there is some degree of freedom.

In my case, it is not possible to recognise the degrees of freedom without considering the algebra behind the equations. 

@kaiwas 
The flipping problem has links to the use of nonlinear polynomials in the constraints. Adding additional quadratic equations in most cases won't result in a stable system. Also metrical or coordinate information may prevent the right solution. Therefore I suggested non-metric topological constraints.

Another solution could be to have weak constraints, e.g. forces or rubber bands or something similar. 

P.S.: Rather than marking unrelated issues as duplicates, I suggest to open a meta-issue collecting references to all the different solver problems.

switch87, 2026-10-17:
Same from the scripting side on 1.0.2 (Gentoo/Linux): `SketchObject` exposes no DOF counter at all (`getDoF()` / `getSolverDoF()` / `LastDoF` all missing), only the `FullyConstrained` boolean — so a script cannot tell an under-determined sketch from a solved one, which is how this bug reaches non-interactive use too. Details and a minimal repro in #15859.