FreeCAD/FreeCAD #15852: Pattern with holes not rendered properly in Part Design

osuchw, 2024-08-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Seems like a regression.  The attached file displays correctly in 0.21.2 but in a weekly version the hole is shown as a pad and not repeating. 
To see the result open the attached file in the weekly version and hit Refresh button (F5).
Creating similar part "from scratch" in the weekly version also does not work.

### Full version info

```shell
>>> sys.version
'3.11.9 | packaged by conda-forge | (main, Apr 19 2024, 18:27:10) [MSC v.1938 64 bit (AMD64)]'
>>> App.Version()
['0', '22', '0', '38429 (Git)', 'https://github.com/FreeCAD/FreeCAD main', '2024/08/09 01:34:28', 'main', '362cf1c80edb89957d81c0dbe6e50ce35daaa6db']
```

### Subproject(s) affected?

PartDesign

### Anything else?

[test-pattern.zip](https://github.com/user-attachments/files/16570440/test-pattern.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

osuchw, 2024-08-10:
Other patterns behave similarly with holes.  Patterns created from pads work as expected

maxwxyz, 2024-08-10:
Please add full about info.
Could you record a macro when recreating and upload it here or paste the code?

osuchw, 2024-08-11:
Version Info:

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * CurvedShapes 1.0.9
  * Curves 0.6.43
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * MeshRemodel 1.9.23
  * QuickMeasure 2022.10.28
  * Silk 0.1.5
  * ThreadProfile 1.92.0

osuchw, 2024-08-11:
The recorded macro was failing with an error :-(
Here is manually edited version that works but only in the weekly.  Apparently some attributes have been renamed since  0.21.2 has been released.

```
# -*- coding: utf-8 -*-
import FreeCAD
import PartDesign
import PartDesignGui
import Sketcher

doc = App.newDocument()
body = doc.addObject('PartDesign::Body','Body')

sketch = body.newObject('Sketcher::SketchObject','Sketch')
sketch.AttachmentSupport = (doc.getObject('XY_Plane'), [''])
sketch.MapMode = 'FlatFace'

geoList = []
geoList.append(Part.LineSegment(App.Vector(-55.782860, 23.364756, 0.000000),App.Vector(-55.782860, -26.635244, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(-55.782860, -26.635244, 0.000000),App.Vector(44.217140, -26.635244, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(44.217140, -26.635244, 0.000000),App.Vector(44.217140, 23.364756, 0.000000)))
geoList.append(Part.LineSegment(App.Vector(44.217140, 23.364756, 0.000000),App.Vector(-55.782860, 23.364756, 0.000000)))
sketch.addGeometry(geoList,False)
del geoList

constraintList = []
constraintList.append(Sketcher.Constraint('Coincident', 0, 2, 1, 1))
constraintList.append(Sketcher.Constraint('Coincident', 1, 2, 2, 1))
constraintList.append(Sketcher.Constraint('Coincident', 2, 2, 3, 1))
constraintList.append(Sketcher.Constraint('Coincident', 3, 2, 0, 1))
constraintList.append(Sketcher.Constraint('Vertical', 0))
constraintList.append(Sketcher.Constraint('Vertical', 2))
constraintList.append(Sketcher.Constraint('Horizontal', 1))
constraintList.append(Sketcher.Constraint('Horizontal', 3))
sketch.addConstraint(constraintList)
del constraintList

sketch.addConstraint(Sketcher.Constraint('Distance',0,1,2,2,100.000000)) 
sketch.addConstraint(Sketcher.Constraint('Distance',1,1,3,2,50.000000)) 

pad = doc.getObject('Body').newObject('PartDesign::Pad','Pad')
pad.Profile = (sketch, ['',])
pad.Length = 10
pad.ReferenceAxis = (sketch,['N_Axis'])
sketch.Visibility = False

pad.Length = 10.000000
pad.TaperAngle = 0.000000
pad.UseCustomVector = 0
pad.Direction = (0, 0, 1)
pad.ReferenceAxis = (sketch, ['N_Axis'])
pad.AlongSketchNormal = 1
pad.Type = 0
pad.UpToFace = None
pad.Reversed = 0
pad.Midplane = 0
pad.Offset = 0

sketch_01 = body.newObject('Sketcher::SketchObject','Sketch001')
sketch_01.AttachmentSupport = (doc.getObject('XY_Plane'),[''])
sketch_01.MapMode = 'FlatFace'

geoList = []
geoList.append(Part.Circle(App.Vector(-40.128689, 0.000000, 0.000000), App.Vector(0.000000, 0.000000, 1.000000), 20.000000))
sketch_01.addGeometry(geoList,False)
del geoList

constraintList = []
sketch_01.addConstraint(Sketcher.Constraint('Diameter',0,20.000000)) 
sketch_01.addConstraint(Sketcher.Constraint('PointOnObject', 0, 3, -1))

doc.recompute()

pocket = body.newObject('PartDesign::Pocket','Pocket')
pocket.Profile = (sketch_01, ['',])
pocket.Length = 5
pocket.ReferenceAxis = (sketch_01,['N_Axis'])
sketch_01.Visibility = False

pocket.Length = 20.000000
pocket.TaperAngle = 0.000000
pocket.UseCustomVector = 0
pocket.Direction = (0, 0, -1)
pocket.ReferenceAxis = (sketch_01, ['N_Axis'])
pocket.AlongSketchNormal = 1
pocket.Type = 0
pocket.UpToFace = None
pocket.Reversed = 1
pocket.Midplane = 0
pocket.Offset = 0

pocket.Visibility = False
pad.Visibility = False
sketch_01.Visibility = False

pattern = body.newObject('PartDesign::LinearPattern','LinearPattern')
pattern.Originals = [pocket,]
pattern.Direction = (sketch_01, ['H_Axis'])
pattern.Length = 80
pattern.Occurrences = 2
pattern.Visibility = True
pattern.Reversed = 0
pattern.Mode = 0
body.Tip = pattern

doc.recompute()
```

maxwxyz, 2024-08-12:
@bgbsww could this be TNP related?

maxwxyz, 2024-08-12:
ok so let's link all issues to the PR then: #15857 #15852 #15856

closed 2024-08-13 by chennes