FreeCAD/FreeCAD #15856: Polar Pattern fails in 0.22

ShayGus, 2024-08-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Polar pattern fails to create in 0.22
Confirmed working as expected in 0.21.2
`# -*- coding: utf-8 -*-

[Polar_Pattern_Issue.FCStd.gz](https://github.com/user-attachments/files/16571792/Polar_Pattern_Issue.FCStd.gz)
+++++++++++++++++++++++++++++++++++++++++++++++++
import FreeCAD
import PartDesign

# Gui.runCommand('Std_DlgMacroRecord',0)
# Gui.Selection.addSelection('Lesson13','Body','Pocket.')
# Gui.Selection.addSelection('Lesson13','Body','Pocket001.')
### Begin command PartDesign_PolarPattern
# Gui.getDocument('Lesson13').ActiveView.setActiveObject('pdbody',App.getDocument('Lesson13').getObject('Body'),'')
App.getDocument('Lesson13').getObject('Body').newObject('PartDesign::PolarPattern','PolarPattern')
App.ActiveDocument.recompute()
App.getDocument('Lesson13').getObject('PolarPattern').Originals = [App.getDocument('Lesson13').getObject('Pocket'),App.getDocument('Lesson13').getObject('Pocket001'),]
App.getDocument('Lesson13').getObject('PolarPattern').Axis = (App.getDocument('Lesson13').getObject('Sketch001'),['N_Axis'])
App.getDocument('Lesson13').getObject('PolarPattern').Angle = 360
App.getDocument('Lesson13').getObject('PolarPattern').Occurrences = 2
App.ActiveDocument.recompute()
# App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.ShapeAppearance=getattr(App.getDocument('Lesson13').getObject('Pocket001').getLinkedObject(True).ViewObject,'ShapeAppearance',App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.ShapeAppearance)
# App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.LineColor=getattr(App.getDocument('Lesson13').getObject('Pocket001').getLinkedObject(True).ViewObject,'LineColor',App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.LineColor)
# App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.PointColor=getattr(App.getDocument('Lesson13').getObject('Pocket001').getLinkedObject(True).ViewObject,'PointColor',App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.PointColor)
# App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.Transparency=getattr(App.getDocument('Lesson13').getObject('Pocket001').getLinkedObject(True).ViewObject,'Transparency',App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.Transparency)
# App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.DisplayMode=getattr(App.getDocument('Lesson13').getObject('Pocket001').getLinkedObject(True).ViewObject,'DisplayMode',App.getDocument('Lesson13').getObject('PolarPattern').ViewObject.DisplayMode)
# Gui.getDocument('Lesson13').setEdit(App.getDocument('Lesson13').getObject('Body'), 0, 'PolarPattern.')
# Gui.Selection.clearSelection()
App.getDocument('Lesson13').getObject('Body').Tip = App.getDocument('Lesson13').getObject('PolarPattern')
App.ActiveDocument.recompute()
### End command PartDesign_PolarPattern
# Gui.Selection.clearSelection()
App.getDocument('Lesson13').getObject('PolarPattern').Visibility = True
App.getDocument('Lesson13').getObject('PolarPattern').Axis = (App.getDocument('Lesson13').getObject('Sketch001'), ['N_Axis'])
App.getDocument('Lesson13').getObject('PolarPattern').Reversed = 0
App.getDocument('Lesson13').getObject('PolarPattern').Mode = 0
App.getDocument('Lesson13').getObject('PolarPattern').Angle = 360.000000
App.getDocument('Lesson13').getObject('PolarPattern').Occurrences = 2
App.getDocument('Lesson13').recompute()
App.getDocument('Lesson13').getObject('Pocket001').Visibility = False
# Gui.getDocument('Lesson13').resetEdit()
### Begin command Std_Save
# Gui.SendMsgToActiveView("Save")
App.getDocument("Lesson13").save()
### End command Std_Save`

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Israel (en_IL)
Installed mods: 
  * Curves 0.6.42
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-11:
could you upload the example file as well?

ShayGus, 2024-08-11:
[Polar_Pattern_Issue.FCStd.gz](https://github.com/user-attachments/files/16572986/Polar_Pattern_Issue.FCStd.gz)

mfraser, 2024-08-11:
This might also affect the mirroring function as I had issues with mirroring a body containing a pad and a pocket earlier using the latest build from the PPA, but in the AppImage it works fine.

```
Version: 0.22.0dev.38334 (Git) AppImage
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
```

tibfel, 2024-08-12:
I have a similar issue when I try to mirror or rotate by 180° the pocket in my file. See attached.
[mirrorIssue.zip](https://github.com/user-attachments/files/16580678/mirrorIssue.zip)

maxwxyz, 2024-08-12:
Is this a duplicate of https://github.com/FreeCAD/FreeCAD/issues/15852 ?

closed 2024-08-13 by chennes