FreeCAD/FreeCAD #15857: Linear pattern breaks the body

nazar-pc, 2024-08-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a body with an element in this that linear pattern is applied to, but this linear pattern causes other part of the body to disappear, the base of the pattern to be shown and the pattern itself not having any effect:

https://github.com/user-attachments/assets/ed9c0c90-ab49-4020-91fe-c330fe26e011

[LinearPatternBodyRenderingBug.zip](https://github.com/user-attachments/files/16573076/LinearPatternBodyRenderingBug.zip)

I'm very confused, it seemed to work before with similar setup, the only thing I did was to re-select edges to include holes.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git) AppImage
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

PartDesign

### Anything else?

May or may not be related to https://github.com/FreeCAD/FreeCAD/issues/15852 and/or https://github.com/FreeCAD/FreeCAD/issues/15856 since I also have holes in my pad.

BTW I have also no idea how I managed to select a sketch outside of the body as the base for extrusion, that is probably a bug as well (though I wish it wasn't, extra binders feel like annoying hacks).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luxwarp, 2024-08-12:
Got the same problem with polar pattern. 

i got this little arm on a cylinder that i want to polar pattern 5 times. 
[image]

Selecting the Pad feature for the arm and then polar pattern, the cylinder disappear and "no support". 
[image]

If i change to transform body it works, however this breaks the model in other places since it will polar pattern the whole body and not only the little arm pad feature.
[image]

nazar-pc, 2024-08-12:
Every time I hit such Part Design issues I just go back to re-doing it with Part workbench and use array tools from Draft workbench. Served me well every time :confused:

maxwxyz, 2024-08-12:
Is this also an issue in 0.21.2 stable version?
@bgbsww I don't think so but could this be TNP related?
@NomAnor could this be related to the last changes on the pattern tool? As well as https://github.com/FreeCAD/FreeCAD/issues/15852 and https://github.com/FreeCAD/FreeCAD/issues/15856

luxwarp, 2024-08-12:
> Is this also an issue in 0.21.2 stable version? @bgbsww I don't think so but could this be TNP related? @NomAnor could this be related to the last changes on the pattern tool? As well as #15852 and #15856

For me all this problems came with the 0.22 version. It works as expected for me in the 0.21.x

bgbsww, 2024-08-12:
#15845 most likely addresses this.  All patterns are transforms, and transforms has a glitch.

luxwarp, 2024-08-12:
@bgbsww Tried the now merged pr and it did not made any difference. Still problem using patterns :)

bgbsww, 2024-08-12:
@luxwarp A couple of things to try:

1) load your file
2) mark the whole thing to recompute
3) Ctrl-r recompute it
4) Go into your pattern, change it to transform Body and the back to transform Tools

There is some weirdness in refresh around that new transform option that does not appear to be a TNP issue; I think that #15845 addresses the TNP portion.   If this doesn't work, can you please post a file?

luxwarp, 2024-08-12:
@bgbsww Thanks for quick response! Tried your little list and it did not work. Tried on both a new document and existing document.

Here you have an example file, try to do a pattern or mirror on the Pad001 feature and you loose the cylinder. :)
[pattern-problem.zip](https://github.com/user-attachments/files/16588931/pattern-problem.zip)

```
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38438 (Git)
Build type: Release
Branch: main
Hash: f846cc7acf8334496ee3305ab636434a56897428
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.5.26
  * FreeCAD-themes 2024.7.24
  * Curves 0.6.43
  * freecad.gears 1.2.0

```

maxwxyz, 2024-08-12:
@nazar-pc @luxwarp could you try with tomorrows weekly build after #15845 was merged: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

bgbsww, 2024-08-12:
38438 should be just after that was merged, so the issue seems legit.

Unfortunately, there are no tests, no examples, and thus it isn't truly clear what the results of the different transform modes should be.  But the legacy tool mode should not break like this.

luxwarp, 2024-08-12:
i also compiled latest main about an hour ago since there was a lot of commits this evening and the problem is still there :).

However, if i don't remember completely wrong the pattern tools worked correctly last week. i will try to compile some older releases of 0.22 and see if i can see when the problem first appeared.

bgbsww, 2024-08-12:
No need to put further effort in here; I have definitively found what I
broke ( wasn't handling makeElementFuse and makeElementCut correctly and
have a patch in process.  There's some other stuff in the method that
really wants to be cleaned up, including a UI issue when you toggle modes
since one of them erases the contents of Originals permanently so you can't
toggle back.  Expect a PR in an hour or so.

BlenderManiac40, 2024-08-13:
If this help in version 0.22 R 38419 Pattern Features worked fine. In recent builds all hell went loose. (Windows Build x64)

NomAnor, 2024-08-13:
> There's some other stuff in the method that really wants to be cleaned up, including a UI issue when you toggle modes since one of them erases the contents of Originals permanently so you can't toggle back.

Not clearing originals might mess with Scale and MultiTransform. They check if it contains entries. I'm not sure what happens in Body mode, when entries to features still exist.

https://github.com/FreeCAD/FreeCAD/blob/aef3bdfe679d7965ee13192e1fc160b18f7e6826/src/Mod/PartDesign/App/FeatureMultiTransform.cpp#L84
https://github.com/FreeCAD/FreeCAD/blob/aef3bdfe679d7965ee13192e1fc160b18f7e6826/src/Mod/PartDesign/App/FeatureScaled.cpp#L72

closed 2024-08-13 by chennes