FreeCAD/FreeCAD #15860: Freecad segfaults on when splitting an ellipsis' edge twice

gnurbs, 2024-08-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

MWE:
- create a document, part design, body, sketch on XY plane
- create an ellipsis on centre point, major radius, point with the centre point constrained to the X axis
- split the ellipsis edge (I split in bottom left quadrant)
- split the ellipsis edge againt (I split in top left quadrant)
- Freecad segfaults

### Full version info

```shell
[code]
OS: Debian GNU/Linux trixie/sid (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.
Build type: Release
Python 3.12.4, Qt 5.15.13, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/Ireland (en_IE)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-08-11:
I don't have 0.21.2 to hand right now but I have been able to reproduce this in 38429 (on an x86 MacBook Pro).

https://github.com/user-attachments/assets/ac861d03-b597-4697-a412-408939a7e09f

(Note there's also a "nan" in the on-view parameters for the minor dimension. Not sure if there's a bug report about this)

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.92.0
  * OpenTheme 2024.7.30
  * fasteners 0.5.25
  * Curves 0.6.42

Thread 0 crash log:

```
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [41693]

VM Region Info: 0 is not in any region.  Bytes before following region: 4500254720
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10c3c7000-10c3e6000    [  124K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Kernel Triage:
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter
VM - (arg = 0x3) mach_vm_allocate_kernel failed within call to vm_map_enter

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Sketcher.so                   	       0x129f409a4 Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>>::getPoint(Part::Geometry const*, Sketcher::PointPos) const + 20
1   Sketcher.so                   	       0x129f4097f Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>>::getPoint(int, Sketcher::PointPos) const + 63
2   SketcherGui.so                	       0x12a702863 SketcherGui::EditModeConstraintCoinManager::processConstraints(Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>> const&) + 771
3   SketcherGui.so                	       0x12a6f96a5 SketcherGui::EditModeCoinManager::processGeometryConstraintsInformationOverlay(Sketcher::GeoListModel<std::__1::unique_ptr<Sketcher::GeometryFacade const, std::__1::default_delete<Sketcher::GeometryFacade const>>> const&, bool) + 389
4   SketcherGui.so                	       0x12a720657 SketcherGui::ViewProviderSketch::draw(bool, bool) + 167
5   SketcherGui.so                	       0x12a7293e7 SketcherGui::ViewProviderSketch::slotSolverUpdate() + 295
6   Sketcher.so                   	       0x129e10d67 boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type>, std::__1::__list_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (), boost::function<void ()>>, boost::signals2::mutex>>, void*>, boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int>>, boost::signals2::slot<void (), boost::function<void ()>>, boost::signals2::mutex>>::dereference() const + 71
7   Sketcher.so                   	       0x129e108c1 boost::signals2::detail::signal_impl<void (), boost::signals2::optional_last_value<void>, int, std::__1::less<int>, boost::function<void ()>, boost::function<void (boost::signals2::connection const&)>, boost::signals2::mutex>::operator()() + 657
8   Sketcher.so                   	       0x129daed2a Sketcher::SketchObject::solve(bool) + 938
9   SketcherGui.so                	       0x12a57c01e SketcherGui::tryAutoRecompute(Sketcher::SketchObject*, bool&) + 126
10  SketcherGui.so                	       0x12a57c0af SketcherGui::tryAutoRecompute(Sketcher::SketchObject*) + 31
11  SketcherGui.so                	       0x12a5fa2eb SketcherGui::DrawSketchHandlerSplitting::releaseButton(Base::Vector2d) + 347
12  SketcherGui.so                	       0x12a71af60 SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) + 1120
13  libFreeCADGui.dylib           	       0x10e3842f7 Gui::ViewProvider::eventCallback(void*, SoEventCallback*) + 503
14  libCoin.80.0.2.dylib          	       0x110cd03b5 SoEventCallback::handleEvent(SoHandleEventAction*) + 197
15  libCoin.80.0.2.dylib          	       0x110abdd89 SoAction::traverse(SoNode*) + 105
16  libCoin.80.0.2.dylib          	       0x110c86ef3 SoChildList::traverse(SoAction*, int, int) + 131
17  libCoin.80.0.2.dylib          	       0x110cda1c5 SoGroup::doAction(SoAction*) + 101
18  libCoin.80.0.2.dylib          	       0x110cf5858 SoSeparator::handleEvent(SoHandleEventAction*) + 40
19  libCoin.80.0.2.dylib          	       0x110abdd89 SoAction::traverse(SoNode*) + 105
[... cut ...]
30  libFreeCADGui.dylib           	       0x10e334ead Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) + 1389
31  libFreeCADGui.dylib           	       0x10e32def4 Gui::NavigationStyle::processEvent(SoEvent const*) + 180
32  libFreeCADGui.dylib           	       0x10e3516df Gui::View3DInventorViewer::processSoEvent(SoEvent const*) + 175
33  libFreeCADGui.dylib           	       0x10e311d07 SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) + 247
34  libQt5Core.5.15.13.dylib      	       0x10fff7273 QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) + 157
```

depthoffocus, 2024-08-11:
Notable extra details:

- the segfault **doesn't** happen for me if the ellipse is drawn with its centre off the X axis and **then** split.
- It does happen if you draw an ellipsis off axis, make the centre coincident with X subsequently, and then split.

Drawing ellipses seems so unreliable in 38429 (in particular with the nan in the on-view parameter) that this is somewhat difficult to test.

gnurbs, 2024-08-11:
> assuming I am following your instructions correctly @gnurbs (though either way it's a crash).

that looks like what I did on my machine. I also didn't have this happening without the constraint, I probably should've mentioned that

mzzx, 2024-09-05:
I'm working on this issue. It seems that the split command copies the `PointOnObject` constraint but fails to set its parameters properly. I'll submit a PR later to fix this.

commit 62460a03 (2024-09-09): Sketcher: transfer constraints on conic's mid point to the newly created conic arc (e.g. ellipse arc)" -m "Fixes #15860" -m "When spliting a sketch curve, constraints on the original curve are transferred to the newly created curve. The original code did not consider the constraints on the center point of an ellipse arc.
commit 4c473a17 (2024-09-09): Sketcher: transfer constraints on conic's mid point to the newly created conic arc (e.g. ellipse arc)" -m "Fixes #15860" -m "When spliting a sketch curve, constraints on the original curve are transferred to the newly created curve. The original code did not consider the constraints on the center point of an ellipse arc.
closed 2024-09-09 by yorikvanhavre