FreeCAD/FreeCAD #15863: Datumplane coordinates are not used in a sketch atached to a datumplane

ChuckInGame, 2024-08-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

you can see here a Sketch022, which is attached to a datumplane "dp_rohr_ablauf_unten"

- Everything which is drawn on this sketch is on the plane level, and not below of it, like you can see in the screenshot.
  You can see the values of the sketch on the left, there is no offset.

I need to close the sketch and reopen it, until the sketch is drawn on the correct position:

[image]

now its correct, I did not changed anything, just closing and reopening the sketch a few times.

The same things happens with Sketch023.

I attached the FCStd zipped to this issue.

I tried to create 2 sketches to swipe on around a wire sketch (Sketch023 around Sketch022), but it does not work but i think, this issues could be a reason for it.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git) AppImage
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Render 2024.8.4
  * ThreadProfile 1.90.0
  * Help 1.0.3
  * freecad.gears 1.2.0
  * AirPlaneDesign 0.4.0
  * 3D_Printing_Tools
  * Assembly4 0.50.13
  * fasteners 0.5.24
  * MeshRemodel 1.9.23
  * pcb 6.2023.1
```

### Subproject(s) affected?

PartDesign

### Anything else?

[watering.zip](https://github.com/user-attachments/files/16575913/watering.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-12:
I cannot reproduce on 
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git)
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7

```

Could you recreate this from scratch (in a simplified file) and record a macro and attach it here?

ChuckInGame, 2024-08-12:
Hi,
many thanks for your time investment 👍 
i tried it, but without luck. There must be a step i'm missing while i try to reproduce it. Give me 1-2 days, may be i can find out, what is nessassary to reproduce this behavior.
I will add a simple file and a record then.
Regards,
chuck

maxwxyz, 2024-08-12:
Ok, thanks!
Please also give us feedback if you cannot reproduce this so this issue could be closed. Please also try it always with the newest weekly version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

ChuckInGame, 2024-08-18:
unable to reproduce in other files, closed

closed 2024-08-18 by ChuckInGame