FreeCAD/FreeCAD #15864: Crashes when using box selection on win11

sametbayici, 2024-08-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Basically unless i manually create sketch when i try to use box selection it crashes i also tried to import a random svg and even extruded it in part workbench but even after that point if i try to use box selection it crashes right away i made a screen recording tried few different things for around 4min but i don't know if it allowed to share a yt video link in here and if it would be any helpfull for diagnosing the cause anyone can lmk if i can further help because this crashing issue not a one time thing it happens everytime i try to use box selection on empty space it happens at every workbench btw atleast i tried it in start, arch, part, part design, sketcher

### Full version info

```shell
version 0.21.2 build 33771
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-08-12:
Sounds like a graphic driver issue.

Perhaps, if you pressed Help → About FreeCAD → Copy to clipboard and pasted the _About_ info here, it would be better.

It would be a thousand times better if you also tried [a weekly bundle](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) and uploaded the report made by _[dxdiag](https://en.wikipedia.org/wiki/DirectX_Diagnostic_Tool)_ and/or _[Speccy](https://en.wikipedia.org/wiki/Speccy)_.

> if it allowed to share a yt video link in here

You [can drag-and-drop an mp4 file](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/attaching-files), just right into the comment box or paste a link to video.

sametbayici, 2024-08-13:
Turns out you were right i tried to force it to run with nvidia gpu on my laptop and it worked fine is there anything that i can do about the driver issue because when i use nvidia graphics on my laptop it drains battery quicker and heats up more and since i like to work while laptop laying on my lap i don't like that...

closed 2024-08-13 by sametbayici