FreeCAD/FreeCAD #15865: Splitting line while preserving constraints doesn't preserve constraints

nazar-pc, 2024-08-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Here is a sketch where one end of the horizontal construction line is in the center of another vertical line (symmetry constraint).

[image]

Using "split edge" tool on construction line preserves length constraint, while symmetry constraint on one of its ends disappears, which is unexpected behavior:

[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38429 (Git) AppImage
Build type: Release
Branch: main
Hash: 362cf1c80edb89957d81c0dbe6e50ce35daaa6db
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-12:
Is this the same as https://github.com/FreeCAD/FreeCAD/issues/12629 or https://github.com/FreeCAD/FreeCAD/issues/12630

maxwxyz, 2024-08-12:
@PaddleStroke FYI