FreeCAD/FreeCAD #15868: Normalize QT-style to Fusion

MisterMakerNL, 2024-08-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

QT has different styles which are a pain to work with as a UI designer, why?
Well because basically you have to check at least [7 styles ](https://doc.qt.io/qt-6/qtquickcontrols-styles.html) on issues.
You also get bug reports of those 7 styles.
And now we have windows 11 that seems to have issues with QT 6, as qt 6 attempt to adapt too windows 11 style. (windows 11 Qt-style? aka number eight?)
So even [QT though](https://www.qt.io/blog/2012/10/30/cleaning-up-styles-in-qt5-and-adding-fusion) this was ridiculous and added 1 style to replace them all called Fusion.
And there fore I think it is a good idea to move Freecad too Fusion too, so Freecad looks the same on every OS.
Also it saves on development time.
For windows 11 it is probably wise to do this sooner than later since it breaks in QT 6.

### Full version info

```shell
NA
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-08-12:
If you are a power user who wants to use his own QT style this is still possible but then your own your own or the others who want to use that style.

krushia, 2024-08-15:
You provide links for two different things: Qt Quick and QStyle.  Which is this bug report about?

obelisk79, 2024-08-15:
This is certainly about QStyle. I generally agree that this should be standardized, however it's worth noting that this essentially will ensure FreeCAD does not follow a native UI style to the host operating system. I don't necessarily think that is bad, but it is definitely important to be aware of. Particularly for Mac users.

MisterMakerNL, 2024-08-15:
> You provide links for two different things: Qt Quick and QStyle. Which is this bug report about?

Qstyle availability is OS depended, as I understand you can even make your own.
And so there isn't really a complete list, and the list although talking about qtquick is I think similar.

krushia, 2024-08-15:
Linux users (myself) will absolutely hate it if our configured QStyle is ignored. I'm not as concerned about Qt Quick, but we usually have that themed as well.

MisterMakerNL, 2024-08-15:
Hate it is a strong word, did you even try out Fusion?
What kind of difference do you see?

krushia, 2024-08-15:
This isn't a "power user" request. This is how desktops work. Users pick the widget style, the colors, the icons, the fonts, the cursors, the language, etc. These are **_configured once_** by the user and applications are expected to abide by the request.

If a newly installed FreeCAD did anything to override these settings, that would be a bug.

You really do not need to control these things. You just pick the type of widgets (combo/spinbox/textarea/...) needed and group them in a container. Qt applies the proper theme to those all by itself.

[image]

obelisk79, 2024-08-16:
You presume too much about how many Linux users customize their desktops to that degree. Regardless of that sort of trivia, *most* FreeCAD stylesheets override much of how those styles look by default anyway according to your screenshot. I use fusion style, but with my themes, everything more closely resembles breeze (again, based on your screenshot)

This change also would impact windows and Mac users in similar ways to Linux. Linux users are just used to having more choice than the alternatives.

All this being said, it isn't some attempt to just foist a style choice into Linux users, rather address inconsistencies and problematic behavior exposed by supporting so many styles, operating systems, and Qt draw styles. Also, not even every Linux DE uses Qt so inconsistency between native/non-native widgets is going to be a problem no matter what.

MisterMakerNL, 2024-08-16:
We supply Freecad to 3 different OS, why wouldn't we make it look the same in every OS? WE can provide same support on every OS.
It's a CAD program, it has to do CAD stuff good the rest isn't important..
Further more I have more Linux computers than Windows and I have never been in the application style setting.
Also changing the handle of the slider from round to square is just a lot of stuff for such a trivial thing. 
99% of users wouldn't notice nor care about the shape of your loading bar and slider.
So I agree with Oblisk I am very sure 99% of Freecad users will not notice and will not care if we do this.
Only as Kadett once stated that Mac users are probably the only one that will notice it as the Fusion style is a bit different as the mac style.

MisterMakerNL, 2024-08-25:
Anyone against just setting the theme's to Qstyle fusion while leaving classic as how it is?

MisterMakerNL, 2024-08-30:
This article even says we should use fusion on Windows 11 as they did not do anything with the Qstyle windows, leaving us hanging with a broken Vista style.
[image]

closed 2024-09-02 by yorikvanhavre